chipweinberger/flutter_blue_plus #516: [Question]: <iOS> After connecting once, the device can no longer be searched

barbara012, 2023-08-25:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.6

### What OS?

iOS

### OS Version

iOS 16.4.1

### Ask your question

Scan the device for the first time, after connecting, and execute the command (the code logic is that every time a command is sent, the device will disconnect the device after it responds normally).
At this time, if I scan again (whether I hot reload or rebuild the app), I can't find the device.
I changed to another Apple mobile phone for real machine debugging. After repeating the above steps, it still had the same result, and the device could not be searched.
Then I downloaded a three-party app (nRF connect) in the app store, it can search for the device, and I can see that the device is in a connectable state

### Logs

```shell
no log
```

chipweinberger, 2023-08-25:
I've never seen this issue before. Not sure what the problem could be. lmk if you fix it,

Brinfotech1407, 2023-08-25:
@barbara012 @chipweinberger 

facing the same issue 
After scanning and finding device and connection, and execute the command once the job is done needs to disconnect device
Now for the subsequent job needs to scan again this time it is not appearing on scan result

is there any way we can destroy the current instance so next time when i scan it should start as a fresh?

chipweinberger, 2023-08-25:
you guys will need to debug it yourself.

we call scan, and then we just pass all the results from the OS to the user

closed 2023-08-29 by barbara012

chipweinberger, 2023-08-29:
@barbara012 can you give more details? how does that cause this problem?