chipweinberger/flutter_blue_plus #517: [Question]: Error when executing setNotifyValue nor current process has android.permission.BLUETOOTH_PRIVILEGED

AlexQiulin, 2023-08-25:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

android 13

### Ask your question

Error when executing setNotifyValue nor current process has android.permission.BLUETOOTH_PRIVILEGED，

### Logs

```shell
D/BluetoothGatt( 3293): connect() - device: E8:D0:FA:2A:B6:36, auto: false, eattSupport: false
D/BluetoothGatt( 3293): registerApp()
D/BluetoothGatt( 3293): registerApp() - UUID=093c5558-e375-4762-98b8-8d735fa3c1c3
D/BluetoothGatt( 3293): onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt( 3293): onClientConnectionState() - status=0 clientIf=8 device=E8:D0:FA:2A:B6:36
D/BluetoothGatt( 3293): onConfigureMTU() - Device=E8:D0:FA:2A:B6:36 mtu=192 status=0
D/BluetoothGatt( 3293): discoverServices() - device: E8:D0:FA:2A:B6:36
D/BluetoothGatt( 3293): onConnectionUpdated() - Device=E8:D0:FA:2A:B6:36 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 3293): onSearchComplete() = Device=E8:D0:FA:2A:B6:36 Status=0
D/BluetoothGatt( 3293): setCharacteristicNotification() - uuid: 0000ffa1-0000-1000-8000-00805f9b34fb enable: true
E/flutter ( 3293): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.SecurityException: Need BLUETOOTH PRIVILEGED permission: Neither user 10784 nor current process has android.permission.BLUETOOTH_PRIVILEGED., java.lang.SecurityException: Need BLUETOOTH PRIVILEGED permission: Neither user 10784 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
E/flutter ( 3293): 	at android.app.ContextImpl.enforce(ContextImpl.java:2250)
E/flutter ( 3293): 	at android.app.ContextImpl.enforceCallingOrSelfPermission(ContextImpl.java:2278)
E/flutter ( 3293): 	at android.content.ContextWrapper.enforceCallingOrSelfPermission(ContextWrapper.java:958)
E/flutter ( 3293): 	at com.android.bluetooth.Utils.enforceBluetoothPrivilegedPermission(Utils.java:422)
E/flutter ( 3293): 	at com.android.bluetooth.gatt.GattService.permissionCheck(GattService.java:575)
E/flutter ( 3293): 	at com.android.bluetooth.gatt.GattService.registerForNotification(GattService.java:4606)
E/flutter ( 3293): 	at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1222)
E/flutter ( 3293): 	at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1210)
E/flutter ( 3293): 	at android.bluetooth.IBluetoothGatt$Stub.onTransact(IBluetoothGatt.java:1086)
E/flutter ( 3293): 	at android.os.Binder.execTransactInternal(Binder.java:1290)
E/flutter ( 3293): 	at android.os.Binder.execTransact(Binder.java:1249)
E/flutter ( 3293): , null)
E/flutter ( 3293): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter ( 3293): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter ( 3293): <asynchronous suspension>
E/flutter ( 3293): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:302:19)
E/flutter ( 3293): <asynchronous suspension>
E/flutter ( 3293): #3      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:204:5)
E/flutter ( 3293): <asynchronous suspension>
E/flutter ( 3293): #4      Ble.openNotify (package:huaxiner/public/ble.dart:117:5)
E/flutter ( 3293): <asynchronous suspension>
E/flutter ( 3293): 
D/BluetoothGatt( 3293): onPhyUpdate() - status=0 address=E8:D0:FA:2A:B6:36 txPhy=2 rxPhy=2
D/BluetoothGatt( 3293): onConnectionUpdated() - Device=E8:D0:FA:2A:B6:36 interval=24 latency=0 timeout=500 status=0
I/com.huaxiner( 3293): This is non sticky GC, maxfree is 8388608 minfree is 2097152
D/BluetoothGatt( 3293): onConnectionUpdated() - Device=E8:D0:FA:2A:B6:36 interval=120 latency=0 timeout=100 status=0
D/BluetoothGatt( 3293): onClientConnectionState() - status=19 clientIf=8 device=E8:D0:FA:2A:B6:36
D/BluetoothGatt( 3293): close()
D/BluetoothGatt( 3293): unregisterApp() - mClientIf=8
```

chipweinberger, 2023-08-25:
try googling the error:

https://stackoverflow.com/questions/19974264/java-lang-securityexception-need-bluetooth-privileged-permission

AlexQiulin, 2023-08-25:
> try googling the error:
> 
> https://stackoverflow.com/questions/19974264/java-lang-securityexception-need-bluetooth-privileged-permission

tq

chipweinberger, 2023-08-31:
did you figure it out? what was the solution?

chipweinberger, 2023-09-02:
closing, assuming my link had the solution

- You can't use BLUETOOTH_ADMIN permission if your app is a third party app (non-system app). To learn more, see Android API: [BLUETOOTH_PRIVILEGED](http://developer.android.com/reference/android/Manifest.permission.html#BLUETOOTH_PRIVILEGED)
- "Took me an hour to realize that the bluetooth on the phone is not enabled. After turning it on, it works as expected."

closed 2023-09-02 by chipweinberger