chipweinberger/flutter_blue_plus #518: [Bug]: Android 13 turnOn method show only the permission window

olegys, 2023-08-25:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.13.1

### What OS?

Android

### OS Version

Android 13

### What happened?

On android 13, when calling the await FlutterBluePlus.turnOn() method, bluetooth does not turn on, only the permission window is shown, and when the permission is accepted, this method freezes and does not return any result
Only after which timer or when the page is reloaded, the FlutterBluePlus.turnOn() method will show the bluetooth enable window

### Logs

```shell
No logs
```

chipweinberger, 2023-08-25:
where in the method does it freeze? set some breakpoints / add print statements

chipweinberger, 2023-08-31:
did you figure it out?

chipweinberger, 2023-09-02:
I'm assuming this is a problem in your own code. More info needed.

closed 2023-09-02 by chipweinberger