chipweinberger/flutter_blue_plus #519: [Question]: why repository example not print, however write success

Gawwad-Tech, 2023-08-28:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

android 10

### Ask your question

The repository example runs without errors. It shows a successful write result when tapping on any characteristic  write button. However, it does not print anything on my Bluetooth 4. X5 thermal printer.

-- Note: The printer functions normally without any issues when using a printer app.

### Logs

```shell
Launching lib\main.dart on Redmi Note 7 in debug mode...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
Connecting to VM Service at ws://127.0.0.1:52507/0S-r_N8je7A=/ws
I/Timeline(20985): Timeline: Activity_launch_request time:15228295 intent:Intent { act=android.content.pm.action.REQUEST_PERMISSIONS pkg=com.google.android.packageinstaller (has extras) }
D/[FBP-Android](20985): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](20985): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android](20985): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(20985): isLeEnabled(): BLE_ON
D/BluetoothLeScanner(20985): could not find callback wrapper
D/[FBP-Android](20985): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](20985): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(20985): isLeEnabled(): BLE_ON
D/BluetoothLeScanner(20985): could not find callback wrapper
D/[FBP-Android](20985): [FBP-Android] onMethodCall: turnOn
I/Timeline(20985): Timeline: Activity_launch_request time:15249910 intent:Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE }
D/[FBP-Android](20985): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](20985): [FBP-Android] OnAdapterStateChanged: turningOn
D/[FBP-Android](20985): [FBP-Android] onMethodCall: stopScan
D/[FBP-Android](20985): [FBP-Android] OnAdapterStateChanged: on
D/[FBP-Android](20985): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(20985): getConnectedDevices
D/[FBP-Android](20985): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(20985): isLeEnabled(): ON
D/BluetoothLeScanner(20985): onScannerRegistered() - status=0 scannerId=9 mScannerId=0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(20985): getConnectedDevices
D/[FBP-Android](20985): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](20985): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(20985): connect() - device: 5D:AC:90:78:D4:D0, auto: false
D/BluetoothGatt(20985): registerApp()
D/BluetoothGatt(20985): registerApp() - UUID=8aaab821-94f0-4b8e-b93e-3634c97df243
D/BluetoothGatt(20985): onClientRegistered() - status=0 clientIf=10
D/BluetoothGatt(20985): onClientConnectionState() - status=0 clientIf=10 device=5D:AC:90:78:D4:D0
D/[FBP-Android](20985): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](20985): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(20985): readRssi() - device: 5D:AC:90:78:D4:D0
D/[FBP-Android](20985): [FBP-Android] onReadRemoteRssi: rssi: -57 status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(20985): discoverServices() - device: 5D:AC:90:78:D4:D0
D/BluetoothGatt(20985): onSearchComplete() = Device=5D:AC:90:78:D4:D0 Status=0
D/[FBP-Android](20985): [FBP-Android] onServicesDiscovered: count: 3 status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(20985): isLeEnabled(): ON
D/[FBP-Android](20985): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicWrite: uuid: 00002a00-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicRead: uuid: 00002a00-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: writeDescriptor
D/[FBP-Android](20985): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/BluetoothGatt(20985): onConnectionUpdated() - Device=5D:AC:90:78:D4:D0 interval=12 latency=0 timeout=600 status=0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: writeDescriptor
D/[FBP-Android](20985): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicWrite: uuid: 0000ae10-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicRead: uuid: 0000ae10-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicWrite: uuid: 0000ae10-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicRead: uuid: 0000ae10-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicRead: uuid: 0000ae10-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicWrite: uuid: 0000ae10-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](20985): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android](20985): [FBP-Android] onCharacteristicRead: uuid: 0000ae10-0000-1000-8000-00805f9b34fb status: 0
```

chipweinberger, 2023-08-29:
impossible to answer the question. I cannot know!

chipweinberger, 2023-08-31:
the example app does not support bluetooth printers.

you would need to modify the example app.

closed 2023-08-31 by chipweinberger