chipweinberger/flutter_blue_plus #520: [Question]: Error on printing qr code

caini1213, 2023-08-29:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.4

### What OS?

Android

### OS Version

11

### Ask your question

Hi, I was trying to print QR code but then error throw as below : 

"The WRITE property is not supported by this BLE characteristic" and
"data longer than mtu allows. dataLength: 456> max: 20"

### Logs

```shell
PlatformException(writeCharacteristic, The WRITE property is not supported by this BLE characteristic, null, null)

PlatformException(writeCharacteristic, data longer than mtu allows. dataLength: 512> max: 20, null, null)
```

chipweinberger, 2023-08-29:
the problem is in your code or, the ble device code.

caini1213, 2023-08-29:
> the problem is in your code or, the ble device code.

can i show u my code to see where should i modify?

chipweinberger, 2023-08-29:
i won't know. i've never used ble printing

caini1213, 2023-08-29:
how to say the problem might in my ble device code ya?

chipweinberger, 2023-09-02:
you can also try write(withoutResponse:true)

closed 2023-09-02 by chipweinberger