chipweinberger/flutter_blue_plus #521: E/ScanRecord(14160): unable to parse scan record: [2, 1, 6, 3, 3, 0, -81, 2, -1, 43, 9, 9, 77, 70, 53, 54, 56, 50, 50, 50]

chikramullah92, 2023-08-30:
### FlutterBluePlus Version

flutter_blue_plus: ^1.14.11

### Flutter Version

3.7.11

### What OS?

Android

### OS Version

13

### What happened?

 why this keep calling when i disconnect all the devices while it is still scanning, but it appears when i disconnect last device

### Logs

```shell
I/flutter (14160): connected devices length: 0
V/studio.transport(14160): Agent handling command (id:143 type:APP_INSPECTION) for pid:14160.
V/studio.transport(14160): Agent handling command (id:144 type:APP_INSPECTION) for pid:14160.
E/ScanRecord(14160): unable to parse scan record: [2, 1, 6, 3, 3, 0, -81, 2, -1, 43, 9, 9, 77, 70, 53, 54, 56, 50, 50, 50]
E/ScanRecord(14160): unable to parse scan record: [2, 1, 6, 3, 3, 0, -81, 2, -1, 43, 9, 9, 77, 70, 53, 54, 56, 50, 50, 50]
E/ScanRecord(14160): unable to parse scan record: [2, 1, 6, 3, 3, 0, -81, 2, -1, 43, 9, 9, 77, 70, 53, 54, 56, 50, 50, 50] and so on until i stop the and re run.
```

chipweinberger, 2023-08-31:
that log comes from android, not FBP.

nothing we can do about it

chipweinberger, 2023-08-31:
more info https://github.com/PhilipsHue/flutter_reactive_ble/issues/521

closed 2023-09-02 by chipweinberger

chikramullah92, 2023-09-02:
Thanks for your quick response.i found the reason. it was due to invalid advertising data.