chipweinberger/flutter_blue_plus #524: [Question]: How can I get the response from device?

expertdev-designer, 2023-08-31:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.5

### What OS?

All

### OS Version

Android 11

### Ask your question

Hii all

Thanks in advance if anyone help me understand this 
I am new to developing the flutter application with Bluetooth as I don't have much knowledge about bluetooth protocol.
With the help of this package, I am able to implement scanning devices integration but could not figure out how I can get the response from the device when I have written the command.

I have checked the example app 
Whenever I write any random byte after that
`  return StreamBuilder<List<int>>(
      stream: widget.characteristic.onValueReceived,
      initialData: widget.characteristic.lastValue,
      builder: (context, snapshot) {
        final List<int>? value = snapshot.data;` 
        
I got the same byte data here 
But how I can get a response like below

```
Device connection password：0x38 38 38 38
APP:0x8f 38 38 38 38 
Device response：0x8f Code 5F 56 32 2E 30

APP:0x8f 38 38 38 11 
Device response：0x8f 45 52 52 4F 52
```

        
        

### Logs

```shell
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android]( 6835): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 6835): isLeEnabled(): ON
D/BluetoothLeScanner( 6835): could not find callback wrapper
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: getConnectedSystemDevices
I/Quality ( 6835): Skipped: false 2 cost 22.978098 refreshRate 0 processName com.mobile.hydur
D/SurfaceComposerClient( 6835): VRR [FRTC] client handle [bufferId:18446744073709551615 framenumber:0] [ffffffff, ffffffff]
D/ViewRootImpl[MainActivity]( 6835):  debugCancelDraw  cancelDraw=false,count = 314,android.view.ViewRootImpl@39bc458
D/SurfaceComposerClient( 6835): VRR [FRTC] client handle [bufferId:18446744073709551615 framenumber:0] [ffffffff, ffffffff]
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter( 6835): isLeEnabled(): ON
D/BluetoothLeScanner( 6835): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: connect
D/BluetoothGatt( 6835): connect() - device: F4:DC:62:72:5A:6F, auto: false, eattSupport: false
D/BluetoothGatt( 6835): registerApp()
D/BluetoothGatt( 6835): registerApp() - UUID=cff7bb02-4994-4d51-a13e-5f3f26924e1e
D/BluetoothGatt( 6835): onClientRegistered() - status=0 clientIf=11
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: discoverServices
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 6835): isLeEnabled(): ON
D/BluetoothGatt( 6835): onClientConnectionState() - status=0 clientIf=11 device=F4:DC:62:72:5A:6F
D/[FBP-Android]( 6835): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 6835): readRssi() - device: F4:DC:62:72:5A:6F
D/[FBP-Android]( 6835): [FBP-Android] onReadRemoteRssi: rssi: -65 status: 0
D/BluetoothGatt( 6835): onConnectionUpdated() - Device=F4:DC:62:72:5A:6F interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 6835): onConnectionUpdated() - Device=F4:DC:62:72:5A:6F interval=36 latency=0 timeout=500 status=0
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt( 6835): discoverServices() - device: F4:DC:62:72:5A:6F
D/BluetoothGatt( 6835): onSearchComplete() = Device=F4:DC:62:72:5A:6F Status=0
D/[FBP-Android]( 6835): [FBP-Android] onServicesDiscovered: count: 3 status: 0
I/flutter ( 6835): ListenValue []
I/flutter ( 6835): ListenValue []
I/flutter ( 6835): ListenValue []
I/flutter ( 6835): ListenValue []
D/BluetoothGatt( 6835): onConnectionUpdated() - Device=F4:DC:62:72:5A:6F interval=126 latency=3 timeout=400 status=0
I/flutter ( 6835): Shadia
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android]( 6835): [FBP-Android] onCharacteristicWrite: uuid: 00002a00-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android]( 6835): [FBP-Android] onCharacteristicRead: uuid: 00002a00-0000-1000-8000-00805f9b34fb status: 0
I/flutter ( 6835): ListenValue [255, 255, 255, 255]
D/[FBP-Android]( 6835): [FBP-Android] onMethodCall: readCharacteristic
D/[FBP-Android]( 6835): [FBP-Android] onCharacteristicRead: uuid: 00002a00-0000-1000-8000-00805f9b34fb status: 0
I/flutter ( 6835): ListenValue [255, 255, 255, 255]
```

chipweinberger, 2023-09-02:
i dont understand the question

expertdev-designer, 2023-09-04:
Hi @chipweinberger 

I want to perform some operations on the diffuser(i.e. device) with the help of the App(with Bluetooth).
but I could not figure out how I could get the response from the diffuser.

`Device connection code：8888　
APP:0x8f 38 38 38 38 code
Password correct, device response：0x8f 43 59 5f 56 33 2e 30 
password wrong, device response：0x8f 45 52 52 4f 52`

whenever i read any data then i get the same byte that I had written.

chipweinberger, 2023-09-04:
i'm sorry I don't know the answer. The README should have all the information you need.

closed 2023-09-04 by expertdev-designer