chipweinberger/flutter_blue_plus #526: [Question]: Error while getting Connected devices

sumanth9977, 2023-08-31:
### FlutterBluePlus Version

flutter_blue_plus 1.14.11

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

12

### Ask your question

Question 1: I am unable to get list of Connected device list.
Question 2: If I am trying to get scanned devices I am getting few devices other then my other Android phone BT.

Q1 : Code to get connected devices.
Future<List<BTDeviceStruct>> getConnectedDevices() async {
  List<BTDeviceStruct> devices = [];
  print("Inside get_connected_devices custom code 1 ");
  List<BluetoothDevice> connectedSystemDevices = await FlutterBluePlus.connectedSystemDevices;
  print("printing $connectedSystemDevices");
  for (var d in connectedSystemDevices) {
    print("Inside get_connected_devices custom code 2");
    await d.connect(); // Must connect *our* app to the device
    final deviceState = await d.connectionState.first;
    if (deviceState == BluetoothConnectionState.connected) {
      final deviceRssi = await d.readRssi();
      devices.add(BTDeviceStruct(
        name: d.localName,
        id: d.remoteId.toString(),
        rssi: deviceRssi,
      ));
    }
  }
  return devices;
}

Q2 Code
Future<List<BTDeviceStruct>> findDevices() async {
  //final flutterBlue = FlutterBluePlus.instance;
  List<BTDeviceStruct> devices = [];
  FlutterBluePlus.scanResults.listen((results) {
    List<ScanResult> scannedDevices = [];
    for (ScanResult r in results) {
      if (r.device.localName.isNotEmpty) {
        scannedDevices.add(r);
      }
    }
    scannedDevices.sort((a, b) => b.rssi.compareTo(a.rssi));
    devices.clear();
    scannedDevices.forEach((deviceResult) {
      devices.add(BTDeviceStruct(
        name: deviceResult.device.localName,
        id: deviceResult.device.remoteId.toString(),
        rssi: deviceResult.rssi,
      ));
    });
  });
  final isScanning = FlutterBluePlus.isScanningNow;
  if (!isScanning) {
    await FlutterBluePlus.startScan(
      timeout: const Duration(seconds: 5),
    );
  }

  return devices;
}

### Logs

```shell
Logs for Question 1
D/BluetoothAdapter( 4070): isLeEnabled(): ON
D/[FBP-Android]( 4070): [FBP-Android] onMethodCall: connect
D/BluetoothGatt( 4070): connect() - device: B8:80:4F:52:E5:7F, auto: false, eattSupport: false
D/BluetoothGatt( 4070): registerApp()
D/BluetoothGatt( 4070): registerApp() - UUID=7353c1ca-3e55-40d9-9d91-6c0aea32367e
D/BluetoothGatt( 4070): onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt( 4070): onClientConnectionState() - status=133 clientIf=8 device=B8:80:4F:52:E5:7F
D/[FBP-Android]( 4070): [FBP-Android] onConnectionStateChange: status: 133 (ANDROID_SPECIFIC_ERROR) newState: disconnected
D/BluetoothGatt( 4070): close()
D/BluetoothGatt( 4070): unregisterApp() - mClientIf=8
I/flutter ( 4070): FlutterBluePlusException: connect: (code: 133) ANDROID_SPECIFIC_ERROR
D/[FBP-Android]( 4070): [FBP-Android] onMethodCall: discoverServices
E/flutter ( 4070): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(discoverServices, device is disconnected, null, null)
E/flutter ( 4070): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter ( 4070): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter ( 4070): <asynchronous suspension>
E/flutter ( 4070): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:302:19)
E/flutter ( 4070): <asynchronous suspension>
E/flutter ( 4070): #3      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:163:7)
E/flutter ( 4070): <asynchronous suspension>
E/flutter ( 4070): #4      connectDevice (package:bluetooth_chat_q_r_code/custom_code/actions/connect_device.dart:24:20)
E/flutter ( 4070): <asynchronous suspension>
E/flutter ( 4070): #5      _HomePageWidgetState.build.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:bluetooth_chat_q_r_code/pages/home_page/home_page_widget.dart:349:67)
E/flutter ( 4070): <asynchronous suspension>
```

chipweinberger, 2023-08-31:
onConnectionStateChange: status: 133 (ANDROID_SPECIFIC_ERROR) newState: disconnected

this is a common error. try googling.

chipweinberger, 2023-09-02:
there's nothing we can do about that error.

closed 2023-09-02 by chipweinberger