chipweinberger/flutter_blue_plus #527: [Bug]:

sqcsabbey, 2023-08-31:
### FlutterBluePlus Version

1.14.9

### Flutter Version

3.13.1

### What OS?

iOS

### OS Version

iOS 15.1

### What happened?

The BluetoothDevice mtu stream keeps reporting 23, but I'm getting messages like:

> PlatformException(writeCharacteristic, data is longer than MTU allows. dataLen: 237 > maxDataLen: 182, null, null);

That's from this plugin's Obj-C code:

```
            // check mtu
            int mtu = (int) [self getMtu:peripheral];
            int dataLen = (int) [self convertHexToData:value].length;
            if ((mtu-3) < dataLen) {
                NSString* f = @"data is longer than MTU allows. dataLen: %d > maxDataLen: %d";
                NSString* s = [NSString stringWithFormat:f, dataLen, (mtu-3)];
                result([FlutterError errorWithCode:@"writeDescriptor" message:s details:NULL]);
                return;
            }
```

... so it looks like, for iOS, that this code has one mtu value and the BluetoothDevice has another. Because 23 is the default, I expect the BluetoothDevice mtu variable isn't getting updated correctly. 

### Logs

```shell
PlatformException(writeCharacteristic, data is longer than MTU allows. dataLen: 237 > maxDataLen: 182, null, null);
```

chipweinberger, 2023-08-31:
is "OnMtuChanged" ever seen in your logs?

I did change how the dart mtu code works. you can look at older versions. we used to actively get the mtu from the system, but now we just passively wait for OnMtuChanged. I find it odd if iOS does not report the mtu change.

chipweinberger, 2023-08-31:
something like this 

```
/// The current MTU size in bytes
  Stream<int> get mtu {
    // get initial value from our cache
    int? initialValue = FlutterBluePlus._mtuValues[remoteId]?.mtu;
    if (initialValue == null) {
        // invoke getMtu from system
       initialValue = XXX
    }
    return FlutterBluePlus._methodStream.stream
        .where((m) => m.method == "OnMtuChanged")
        .map((m) => m.arguments)
        .map((args) => BmMtuChangedResponse.fromMap(args))
        .where((p) => p.remoteId == remoteId.str)
        .map((p) => p.mtu)
        .newStreamWithInitialValue(initialValue!);
  }
```

chipweinberger, 2023-08-31:
yes this looks like a real bug in flutter blue plus. 

i'm surprised no one else has noticed this.

in iOS, onMtuChanged is not exposed by iOS. We must actively ask for the mtu from the system. 

I can fix this later next week.

chipweinberger, 2023-08-31:
fixed: 66655d6dd3705d284a80897ada039cc0749500aa

sqcsabbey, 2023-09-01:
Thank you for the fix, @chipweinberger!

A related question, is there any way to tell the difference between the default value and a reading from the peripheral? I'm planning on listening to the mtu stream until the value isn't 23, but this is awkward... I'd think the stream should only be populated with actual readings, and users of this plugin could implement their own default.

chipweinberger, 2023-09-01:
yes i was thinking about that issue as well.

before i release the next version i'll improve it 

i think the solution is to change the order so that onMtuChanged is called before onConnectionStateChanged

then, you can wait until connection. after connection you can `await device.mtu.first` and it will be the correct value

sqcsabbey, 2023-09-01:
I tried a later patch ([a5c2396](https://github.com/boskokg/flutter_blue_plus/commit/a5c239659fe89fcc9d1a71746a6d4193423b9aab)), and I can see the new code is getting called, but I'm still only getting a single value of 23 from the mtu stream on my iPhone 6s.

chipweinberger, 2023-09-02:
odd. at what point does the mtu change to a larger value?

sqcsabbey, 2023-09-02:
> odd. at what point does the actual mtu change to a larger value?

Not sure, I'm going to go into the office on Tuesday and debug all of the BT traffic to figure it out.

sqcsabbey, 2023-09-05:
Here's the request to our peripheral:
[image]

... and the response from our peripheral:
[image]

It looks like the value of 23 is coming from this plugin, not the peripheral. 

Here's a discussion on the [Apple forums](https://developer.apple.com/forums/thread/73871) about how to get the MTU size, and whether the onMtuChanged callback is actually called.

sqcsabbey, 2023-09-05:
@chipweinberger 

> i'm surprised no one else has noticed this.

It looks like the error is relatively new (`PlatformException(writeCharacteristic, data is longer than MTU allows. dataLen: 237 > maxDataLen: 182, null, null);`); I think this was failing silently before.

chipweinberger, 2023-09-05:
my main question is when will calling `getMtu` return the correct value?

currently i call it in the onConnectes handler. is that too soon?

chipweinberger, 2023-09-05:
perhaps you can try adding a separate thread that continually calls `getMtu`

and then we can figure out when it changes

sqcsabbey, 2023-09-05:
I added 

```
[NSTimer scheduledTimerWithTimeInterval:.01
    target:self
    selector:@selector(logMtu:) 
    userInfo:@{@"peripheral":peripheral,}
    repeats:YES];
```

to `didConnectPeripheral`, with logMtu of

```
- (void)logMtu:(NSTimer *)timer {
    CBPeripheral * peripheral = timer.userInfo[@"peripheral"];
    NSLog( @"[FBP-iOS] logMtu %d", [self getMtu:peripheral]);
}
``` 

... and it looks like the value is ready in .06s for my phone and peripheral:

> [FBP-iOS] didConnectPeripheral
> [FBP-iOS] logMtu 23
> [FBP-iOS] handleMethodCall: discoverServices
> [FBP-iOS] logMtu 23
> [FBP-iOS] logMtu 23
> [FBP-iOS] logMtu 23
> [FBP-iOS] logMtu 23
> [FBP-iOS] logMtu 185

chipweinberger, 2023-09-05:
> I think this was failing silently before.

on iOS, the platform will silently split your payload into multiple packets.

that is bad for performance, so i've made it an error, and provided a separate function (`writeLarge`) in the README if you want auto splitting behavior.

chipweinberger, 2023-09-05:
> it looks like the value is ready in .06s

Thanks for testing!!

This is tricky. How should we handle this?

perhaps we can
- add a timer that checks for mtu changes (just like you did)
- also alter the getMtu function to check for mtu changes 

when the mtu changes:
- push to stream

when flutter blue plus ios code is initialized: 
- create timer

chipweinberger, 2023-09-05:
I've pushed a fix to master: 5dc29c4ef25b3e1b7c7615d41070c043da834baa

let me know if it works for you.

sqcsabbey, 2023-09-06:
Is there a similar problem for large reads? We are getting some truncated data on the iPhone 6S; e.g. the length sent with the data says there should be 212 bytes, but we're now only receiving 182 bytes.

chipweinberger, 2023-09-06:
> assuming that this MTU read gets passed along to the peripheral

no it does not get passed on. it is just reading a cached value.

canceling the timer could be nice but more complicated to support multiple devices. also a device may only support 23 MTU, so we'd need some timeout (maybe 30 seconds?). Overall kinda complicated to save what i think is a very small amount of perf. But i wish there was a simpler solution than a timer.

> Is there a similar problem for large reads? We are getting some truncated data on the iPhone 6S; e.g. the length sent with the data says there should be 212 bytes, but we're now only receiving 182 bytes.

I would think this is a problem with your device firmware. 

when you say *now* what do you mean? did it work on 1.4.0?

chipweinberger, 2023-09-06:
[image]

[image]

It seems `writeLarge` may not be work with subsequent reads if data > mtu. 

I could remove these checks. But we need to make sure android operates the same way.

But perhaps there is some other way to fix this issue.

chipweinberger, 2023-09-06:
according to chatGPT, reading a characteristic on Android when data > MTU truncates to MTU length. There seems to be no way to get the rest of the data.

If this is true, then we should make sure iOS operates the same way.

chipweinberger, 2023-09-06:
in android, it seems the maximum write length is 512, hardcoded (note: GATT_MAX_ATTR_LEN == 512)

https://cs.android.com/android/platform/superproject/main/+/main:packages/modules/Bluetooth/system/stack/gatt/gatt_sr.cc

[image]

chipweinberger, 2023-09-06:
fixed in 1.14.13.

open a new issue if you still have problems.

closed 2023-09-06 by chipweinberger

sqcsabbey, 2023-09-06:
> in android, it seems the maximum write length is 512, hardcoded (note: GATT_MAX_ATTR_LEN == 512)

Different Android versions will support different values; I believe Android 14+ have a mtu of [517](https://developer.android.com/about/versions/14/behavior-changes-all#mtu-set-to-517).