chipweinberger/flutter_blue_plus #528: [Bug]: autoConnect:true doesn't reconnect to device after losing connection.

ericaig, 2023-09-01:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.7.11

### What OS?

Android

### OS Version

Android 13

### What happened?

In version `flutter_blue_plus 1.4.0`, after losing a connection with a Bluetooth device, a connection is automatically established whenever the BLE device is in range. But in version `flutter_blue_plus 1.14.11` this doesn't work. Of course, I'm connecting to the device like this `device.connect(autoConnect: true)`. My app is notified when the connection is lost, but never receives any when the device is turned back on or in range.

### Logs

```shell
I/flutter (20720): [FBP] <getConnectedSystemDevices> args: null
I/flutter (20720): [FBP] <getConnectedSystemDevices> result: {devices: []}
I/flutter (20720): [FBP] <getAdapterState> args: null
I/flutter (20720): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter (20720): Scanning for devices...
I/flutter (20720): [FBP] <startScan> args: {service_uuids: [0000180a-0000-1000-8000-00805f9b34fb], mac_addresses: [], allow_duplicates: false, android_scan_mode: 2, android_uses_fine_location: false}
I/flutter (20720): [FBP] <startScan> result: null
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -74, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -66, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -64, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -66, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -71, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -69, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -67, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] <stopScan> args: null
I/flutter (20720): Scan completed. Found 1 devices with matching serviceUUID
I/flutter (20720): [FBP] <stopScan> args: null
I/flutter (20720): [FBP] <stopScan> result: null
I/flutter (20720): [FBP] <stopScan> result: null
I/flutter (20720): Empty tag detected
I/flutter (20720): Scanning for devices... [XX:XX:XX:XX:XX:XX]
I/flutter (20720): [FBP] <getConnectedSystemDevices> args: null
I/flutter (20720): [FBP] <getConnectedSystemDevices> result: {devices: []}
I/flutter (20720): [FBP] <startScan> args: {service_uuids: [0000180a-0000-1000-8000-00805f9b34fb], mac_addresses: [], allow_duplicates: false, android_scan_mode: 0, android_uses_fine_location: false}
I/flutter (20720): [FBP] <startScan> result: null
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -62, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -54, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -44, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] [[ OnScanResponse ]] result: {result: {rssi: -59, advertisement_data: {connectable: 1, manufacturer_data: {}, service_data: {}, local_name: XXXX_XXXXXXXXXXXX, service_uuids: [0000180a-0000-1000-8000-00805f9b34fb]}, device: {remote_id: XX:XX:XX:XX:XX:XX, type: 2, local_name: XXXX_XXXXXXXXXXXX}}}
I/flutter (20720): [FBP] <stopScan> args: null
I/flutter (20720): Scan completed. Found 1 peripherals with matching serviceUUID
I/flutter (20720): [FBP] <stopScan> args: null
I/flutter (20720): [FBP] <stopScan> result: null
I/flutter (20720): [FBP] <stopScan> result: null
I/flutter (20720): Found PERIPHERAL:: XX:XX:XX:XX:XX:XX
I/flutter (20720): [FBP] <connect> args: {remote_id: XX:XX:XX:XX:XX:XX, auto_connect: 1}
I/flutter (20720): [FBP] <connect> result: 0
I/flutter (20720): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: XX:XX:XX:XX:XX:XX, connection_state: 1}
I/flutter (20720): [FBP] <discoverServices> args: XX:XX:XX:XX:XX:XX
I/flutter (20720): [FBP] <discoverServices> result: null
I/flutter (20720): [FBP] [[ OnDiscoverServicesResult ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: XX:XX:XX:XX:XX:XX, error_code: 0, services: [{included_services: [], characteristics: [{secondary_service_uuid: null, descriptors: [], service_uuid: 00001800-0000-1000-8000-00805f9b34fb, remote_id: XX:XX:XX:XX:XX:XX, characteristic_uuid: 00002a00-0000-1000-8000-00805f9b34fb, value: , properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 1, notify: 0}}, {secondary_service_uuid: null, descriptors: [], service_uuid: 00001800-0000-1000-8000-00805f9b34fb, remote_id: XX:XX:XX:XX:XX:XX, characteristic_uuid: 00002a01-0000-1000-8000-00805f9b34fb, value: , properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0,
I/flutter (20720): [FBP] <setNotification> args: {remote_id: XX:XX:XX:XX:XX:XX, service_uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, secondary_service_uuid: null, characteristic_uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, enable: true}
I/flutter (20720): Connected to device
I/flutter (20720): [XX:XX:XX:XX:XX:XX] Connection state changed to {-connected}-
I/flutter (20720): [FBP] <setNotification> result: null
I/flutter (20720): [FBP] [[ OnDescriptorResponse ]] result: {secondary_service_uuid: null, error_string: GATT_SUCCESS, service_uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca9e, success: 1, remote_id: XX:XX:XX:XX:XX:XX, descriptor_uuid: 00002902-0000-1000-8000-00805f9b34fb, error_code: 0, type: 1, characteristic_uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: 0100}
I/flutter (20720): [FBP] [[ OnBondStateChanged ]] result: {bond_failed: false, bond_state: 1, bond_lost: false, remote_id: XX:XX:XX:XX:XX:XX}
I/flutter (20720): App Lifecycle state: inactive
I/flutter (20720): App Lifecycle state: resumed
I/flutter (20720): App Lifecycle state: inactive
I/flutter (20720): App Lifecycle state: resumed
I/flutter (20720): [FBP] [[ OnBondStateChanged ]] result: {bond_failed: false, bond_state: 2, bond_lost: false, remote_id: XX:XX:XX:XX:XX:XX}
[... cut ...]
I/flutter (20720): [XX:XX:XX:XX:XX:XX] Connection state changed to {-disconnected}-
I/flutter (20720): Automatically disconnected from device... Uh hooooooooooo
I/flutter (20720): [FBP] <getConnectedSystemDevices> args: null
I/flutter (20720): [FBP] <getConnectedSystemDevices> result: {devices: []}
I/flutter (20720): [FBP] <getConnectedSystemDevices> args: null
I/flutter (20720): [FBP] <getConnectedSystemDevices> result: {devices: []}
I/flutter (20720): [FBP] <getConnectedSystemDevices> args: null
I/flutter (20720): [FBP] <getConnectedSystemDevices> result: {devices: []}
```

chipweinberger, 2023-09-02:
thanks for filing. I'm not sure the problem to be honest.

MartinXBcn, 2023-09-03:
Same for me: in older versions of flutter_blue_plus (I can not say until which version) the auto-re-connect worked, too, and now it does work anymore.

chipweinberger, 2023-09-04:
I probably will not fix this issue myself.

if you figure it out, open a PR please.

MartinXBcn, 2023-09-04:
I think I have figured out why it is not reconnecting automatically anymore.

There is a conceptual change how to react on ``BluetoothProfile.STATE_DISCONNECTED`` in the Gatt-callback: you destroy every time there is a "disconnection" all connection-data by calling ``gatt.close()`` although the connection was "only" lost for example because being out of range. In this case a later re-connection is not possible. The "old" version (until version 1.4.0) only called  ``gatt.close()`` when the disconnection was initiated by the user but kept the connection-data in all other cases of disconnection.

@chipweinberger Can you confirm my analysis? Do you see any problem to change back to the previous concept of calling ``gatt.close()`` only when disconnect was initiated by the user?

chipweinberger, 2023-09-04:
thanks for your analysis

we need to call close to free up resources. After 6 allocations, new connections will no longer be possible. 

that said, i could change the code to not call `gatt.close` if autoconnect is enabled for that connection.

perhaps you can try that and let me know if it works.

ericaig, 2023-09-05:
Well done @MartinXBcn. I'm sorry I can't help you guys on this. I don't have too much knowledge with native code. 
Ets el rei Martin!

MartinXBcn, 2023-09-05:
> thanks for your analysis
> 
> yes we changed the behavior of `gatt.close` to free up resources. Most phones only have enough resources for 6 connections, and will deny new connections. It is best to free resources automatically when the connection is lost.
> 
> that said, we could change the code to not call `gatt.close` if autoconnect is enabled for that connection.
> 
> in this case, the user will have to manually call disconnect to free up resources.
> 
> perhaps you can try that and let me know if it works.

That is a good idea and even an improvement to the "old" concept which kept the connection-data unnecessarily even without "autoConnect" activated.

I will create a PR.

chipweinberger, 2023-09-05:
fixed in master: 3d3bd1063cd778d17d2b0dec3e57a1f11d3dd5fe

chipweinberger, 2023-09-05:
should be fixed in 1.14.12

closed 2023-09-05 by chipweinberger

ericaig, 2023-09-06:
Thanks, @MartinXBcn @chipweinberger