chipweinberger/flutter_blue_plus #529: [Bug]: Timestamp of Scan Result Not Updating in iOS Version

leelai, 2023-09-01:
### FlutterBluePlus Version

Dart SDK 3.1.0-11.0.dev

### Flutter Version

3.10.0-6.0.pre.27

### What OS?

iOS

### OS Version

12

### What happened?

Observe that the timestamp of the scan results remains static and doesn't update as expected.

class ScanResult {
  final BluetoothDevice device;
  final AdvertisementData advertisementData;
  final int rssi;
  final DateTime timeStamp; <--here.

### Logs

```shell
the example of the project.
```

leelai, 2023-09-01:
and i found example will updated the timestamp of all my apple devices(ipad, macbook..), but non-apple devices will not.

leelai, 2023-09-01:
I need to keep pressing scan button in example app to make timestamp of scan result updated.

chipweinberger, 2023-09-02:
does allowDuplicates fix it?

leelai, 2023-09-02:
allowDuplicates works. thank you.

closed 2023-09-02 by leelai