chipweinberger/flutter_blue_plus #531: [Bug]: Can't write without response after updating FlutterBluePlus

serge18, 2023-09-01:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.1

### What OS?

Android

### OS Version

12

### What happened?

I am using FlutterBluePlus for printing stickers on Toshiba portable sticker. In case of long label string chunk writing with response took long time (around 3 seconds). In version 1.7.5 I can write without response and it was very fast.
Why this option is disabled in the latest version? 
Error message: "The WRITE_NO_RESPONSE property is not supported by this BLE characteristic"

### Logs

```shell
[INFO] BluetoothService: Connected to device
```

chipweinberger, 2023-09-02:
you generally should not call "no response" on a characteristic that does not advertise support.

I'm not sure what the OS will do in that scenario. 

If you can tell me how the OS handles this scenario on both android and ios, i may add an option to remove this check.

chipweinberger, 2023-09-02:
on ios without response is ignored in this scenario

see https://github.com/capacitor-community/bluetooth-le/issues/104

in other words your withoutResponse flag is not actually working on iOS!

chipweinberger, 2023-09-02:
it looks like in android they ignore your withoutResponse flag as well, but im not sure.

https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/framework/java/android/bluetooth/BluetoothGattCharacteristic.java?q=%09WRITE_TYPE_NO_RESPONSE

```
        if ((mProperties & PROPERTY_WRITE_NO_RESPONSE) != 0) {
            mWriteType = WRITE_TYPE_NO_RESPONSE;
        } else {
            mWriteType = WRITE_TYPE_DEFAULT;
        }
```

chipweinberger, 2023-09-02:
In android it is hard to tell what happens. The source code has many levels of abstraction. This is the lowest level i've found in the code so far:

https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/bta/gatt/bta_gattc_api.cc?q=BTA_GATTC_WriteCharValue&ss=android%2Fplatform%2Fsuperproject

```
void BTA_GATTC_WriteCharValue(uint16_t conn_id, uint16_t handle,
                              tGATT_WRITE_TYPE write_type,
                              std::vector<uint8_t> value,
                              tGATT_AUTH_REQ auth_req,
                              GATT_WRITE_OP_CB callback, void* cb_data) {
  tBTA_GATTC_API_WRITE* p_buf = (tBTA_GATTC_API_WRITE*)osi_calloc(
      sizeof(tBTA_GATTC_API_WRITE) + value.size());

  p_buf->hdr.event = BTA_GATTC_API_WRITE_EVT;
  p_buf->hdr.layer_specific = conn_id;
  p_buf->auth_req = auth_req;
  p_buf->handle = handle;
  p_buf->write_type = write_type;
  p_buf->len = value.size();
  p_buf->write_cb = callback;
  p_buf->write_cb_data = cb_data;

  if (value.size() > 0) {
    p_buf->p_value = (uint8_t*)(p_buf + 1);
    memcpy(p_buf->p_value, value.data(), value.size());
  }

  bta_sys_sendmsg(p_buf);
}
```

chipweinberger, 2023-09-02:
In android, it appears that you can actually do a withoutResponse on a characteristic that does not support it:

https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/system/stack/gatt/gatt_cl.cc?q=GATT_WRITE_NO_RSP&ss=android%2Fplatform%2Fsuperproject

It will do either a `GATT_REQ_WRITE` for withResponse, and a `GATT_CMD_WRITE` for withoutResponse.

```
/** GATT write operation */
void gatt_act_write(tGATT_CLCB* p_clcb, uint8_t sec_act) {
  tGATT_TCB& tcb = *p_clcb->p_tcb;

  CHECK(p_clcb->p_attr_buf);
  tGATT_VALUE& attr = *((tGATT_VALUE*)p_clcb->p_attr_buf);

  uint16_t payload_size = gatt_tcb_get_payload_size(tcb, p_clcb->cid);

  switch (p_clcb->op_subtype) {
    case GATT_WRITE_NO_RSP: {
      p_clcb->s_handle = attr.handle;
      uint8_t op_code = (sec_act == GATT_SEC_SIGN_DATA) ? GATT_SIGN_CMD_WRITE
                                                        : GATT_CMD_WRITE;
      tGATT_STATUS rt = gatt_send_write_msg(tcb, p_clcb, op_code, attr.handle,
                                            attr.len, 0, attr.value);
      if (rt != GATT_CMD_STARTED) {
        if (rt != GATT_SUCCESS) {
          LOG(ERROR) << StringPrintf(
              "gatt_act_write() failed op_code=0x%x rt=%d", op_code, rt);
        }
        gatt_end_operation(p_clcb, rt, NULL);
      }
      return;
    }

    case GATT_WRITE: {
      if (attr.len <= (payload_size - GATT_HDR_SIZE)) {
        p_clcb->s_handle = attr.handle;

        tGATT_STATUS rt = gatt_send_write_msg(
            tcb, p_clcb, GATT_REQ_WRITE, attr.handle, attr.len, 0, attr.value);
        if (rt != GATT_SUCCESS && rt != GATT_CMD_STARTED &&
            rt != GATT_CONGESTED) {
          if (rt != GATT_SUCCESS) {
            LOG(ERROR) << StringPrintf(
                "gatt_act_write() failed op_code=0x%x rt=%d", GATT_REQ_WRITE,
                rt);
          }
          gatt_end_operation(p_clcb, rt, NULL);
        }

      } else {
        /* prepare write for long attribute */
        gatt_send_prepare_write(tcb, p_clcb);
      }
      return;
    }

    case GATT_WRITE_PREPARE:
      gatt_send_prepare_write(tcb, p_clcb);
      return;

    default:
      CHECK(false) << "Unknown write type" << p_clcb->op_subtype;
      return;
  }
}
```

chipweinberger, 2023-09-02:
These values are part of the BLE specification

```
  GATT_REQ_WRITE = 0x12,
  GATT_RSP_WRITE = 0x13,
  GATT_CMD_WRITE = 0x52,
```

They represent specific operation codes (opcodes) for GATT procedures.

Here's how they are represented in the BLE specification:

**0x12: ATT_OP_WRITE_REQ**

This is the opcode for the Write Request. A client sends this request to write a value to a GATT characteristic on the server. The client expects a response from the server indicating the success or failure of the write operation.

**0x13: ATT_OP_WRITE_RSP**

This is the opcode for the Write Response. The server sends this response after receiving a Write Request (0x12) from the client. It indicates the success or failure of the write operation.

**0x52: ATT_OP_WRITE_CMD**

This is the opcode for the Write Command. Unlike the Write Request (0x12), the Write Command does not expect a response from the server. It's essentially a "fire-and-forget" method for writing data.

chipweinberger, 2023-09-02:
So in summary.

**android** _allows_ sending a `ATT_OP_WRITE_CMD` (i.e. no response) to a characteristic that does not support it.  There are a few possibilities (the ble specification does specify)
- the ble device may process the write as requested, and not send a response, just like you wanted!
- the ble device may ignore the write completely
- the ble device may treat it as a withResponse and return a response anyway

on **iOS**: sending a withoutResponse to a characteristic that does not support it will silently ignore the write.

**Ideally, the goal of FlutterBluePlus is behave the same way on both platforms. I am not going to remove this check. In fact, I am going to add the same check to iOS =) That way both iOS and Android will return the same error.**

closed 2023-09-02 by chipweinberger