chipweinberger/flutter_blue_plus #532: [Feature]: Retry Android Gatt Connection in Platform Code

cameronkloot, 2023-09-03:
### FlutterBluePlus Version

1.14.9

### Flutter Version

3.7.10

### What OS?

Android

### OS Version

Android 12

### Feature Request

Android Gatt connection often times out after five seconds. This will happen if the device is out of range, but also if the device is close to the phone. If the connection doesn't happen within five seconds, the connection state == disconnected in FlutterBluePlusPlugin.java and the plugin kicks back a ANDROID_SPECIFIC_ERROR, disconnect_reason_code 133:

```
I/flutter (16363): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 133, disconnect_reason_string: ANDROID_SPECIFIC_ERROR, remote_id: CC:F9:57:82:13:CD, connection_state: 0}
```

Relevant existing issues: 
https://github.com/boskokg/flutter_blue_plus/issues/501
https://github.com/boskokg/flutter_blue_plus/issues/526

One of the links in @chipweinberger's comment in #501 suggests retrying the Gatt connection if there's a connection state == disconnected. I was reminded that I had the same retry logic in an older version of my app written in Kotlin. I have solved this problem by adding a retry counter to my connect function:

```
Future connect(BluetoothDevice device, {int retryCount = 0}) async {
  FlutterBluePlus.setLogLevel(LogLevel.verbose, color: true);
  // Ensure scan is fully stopped
  await Future.delayed(const Duration(milliseconds: 500));
  _device = device;
  try {
    await device.connect(); // Default 35 second timeout
  } catch (e) {
    if (e.toString().contains('ANDROID_SPECIFIC_ERROR') && retryCount < 7) {
      return connect(device, retryCount: retryCount + 1);
    } else {
      throw Exception('Connection timeout');
    }
  }
}
```

My app now connects to the device correctly. Sometimes it's within five seconds, and there is no retry. Other times, the app retries a few times. If the device is out of range, the app cycles through the retries and shows a connection timeout, as expected. 

I'm posting this issue so other people can implement the retries in their dart code. I think it would make sense to add this logic to the plugin Java code as well. I could try putting together a pull request, but I'm not sure I can do a very good job. There doesn't seem to be any way to reference the current method call in the onConnectionStateChange method, but I would think we could set class variable "current method" and have a connection retry variable to run the gatt connect on disconnection. I'm also not sure about how the five second disconnection would work for other Android phones. My Galaxy S10 and Galaxy A23 both have this behavior. I can check on other devices in the future. 

: Cameron

### Logs

```shell
D/[FBP-Android](16363): [FBP-Android] onMethodCall: getAdapterState
I/flutter (16363): 2023-09-03 10:17:01.541348 | INFO | Connect to knownDevice: CC:F9:57:82:13:CD: SW4-13CD
I/flutter (16363): 	UpdateRouteState.initState.<fn>/:268
I/flutter (16363): 2023-09-03 10:17:01.543714 | INFO | Connect to device SW4-13CD
I/flutter (16363): 	UpdateRouteState._connect/:44
I/flutter (16363): 2023-09-03 10:17:01.545165 | INFO | Connect with known device SW4-13CD - CC:F9:57:82:13:CD
I/flutter (16363): 	MdBleManager.connectWithKnown/:125
D/[FBP-Android](16363): [FBP-Android] onMethodCall: setLogLevel
I/flutter (16363): [FBP] <connect> args: {remote_id: CC:F9:57:82:13:CD, auto_connect: 0}
D/[FBP-Android](16363): [FBP-Android] onMethodCall: connect
I/BluetoothAdapter(16363): STATE_ON
D/BluetoothGatt(16363): connect() - device: CC:F9:57:82:13:CD, auto: false
I/BluetoothAdapter(16363): isSecureModeEnabled
D/BluetoothGatt(16363): registerApp()
D/BluetoothGatt(16363): registerApp() - UUID=8f591657-117e-4d27-a673-480b3dd46e32
D/BluetoothGatt(16363): onClientRegistered() - status=0 clientIf=13
I/flutter (16363): [FBP] <connect> result: 0
D/BluetoothGatt(16363): onClientConnectionState() - status=133 clientIf=13 device=CC:F9:57:82:13:CD
D/[FBP-Android](16363): [FBP-Android] onConnectionStateChange: status: 133 (ANDROID_SPECIFIC_ERROR) newState: disconnected
D/BluetoothGatt(16363): close()
D/BluetoothGatt(16363): unregisterApp() - mClientIf=13
I/flutter (16363): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 133, disconnect_reason_string: ANDROID_SPECIFIC_ERROR, remote_id: CC:F9:57:82:13:CD, connection_state: 0}
```

chipweinberger, 2023-09-03:
thanks for filing. Perhaps we could add a "retry" argument to `device.connect` in the dart code.

cameronkloot, 2023-09-03:
You're right, it makes more sense to put it in bluetooth_device.dart. Something like this: 

```
final _StreamController<bool> _isDiscoveringServices =
    _StreamController(initialValue: false);

int _retryCount = 0;
```
...
```
Future<void> connect({
  Duration timeout = const Duration(seconds: 35),
  bool autoConnect = false,
  int retryCount = 0,
}) async {
```
...
```
if (response.connectionState == BmConnectionStateEnum.disconnected) {
  if (retryCount < _retryCount) {
    connect(
      timeout: timeout,
      autoConnect: autoConnect,
      retryCount: retryCount += 1,
    );
  } else {
    throw FlutterBluePlusException(_nativeError, "connect",
        response.disconnectReasonCode, response.disconnectReasonString);
  }
}
```

cameronkloot, 2023-09-05:
One potential issue occurred to me: Device is connected, device is disconnected, and the same reference device is connected again. The local retryCount would not be reset to zero. Handling this outside of the package is not a problem because I always get a new device reference before connecting. Not sure the best way to solve if retry is incorporated into bluetooth_device.dart.

chipweinberger, 2023-09-06:
Yes I think this is best for the developer to implement themself.

closed 2023-09-06 by chipweinberger