chipweinberger/flutter_blue_plus #533: [Feature]: macos low version adaptation problem

mengzhidaren, 2023-09-04:
### FlutterBluePlus Version

1.4.11

### Flutter Version

3.13.0

### What OS?

macOS

### OS Version

macos 10.14

### Feature Request

`
s.platform = :osx, '12.00'

s.framework = 'CoreBluetooth'
`
There are still many old macos versions on the market that are only 10.14. The macos version upgrade may not be as fast as ios. Why does the version support on the macos platform start from 12.0? I don’t know much about macos development, but I inquired about CoreBluetooth. Support is low Version, is there any plan to support lower version platforms in the future?

chipweinberger, 2023-09-05:
I'm not sure. I imagine it was set to 12.00 to fix some build error.

You can try setting it lower, I'm sure it will work no problem.

chipweinberger, 2023-09-06:
did you try lowering it?

chipweinberger, 2023-09-06:
fixed in 1.14.16

lowered to `macos 10.11`

closed 2023-09-06 by chipweinberger

mengzhidaren, 2023-09-12:
Many thanks to the author for his contribution