chipweinberger/flutter_blue_plus #536: [Question]: On iOS, after the Bluetooth device's broadcast signal is first scanned and connected to Bluetooth, the device's broadcast signal is no longer scanned (the Bluetooth device is not connected).

wanan76, 2023-09-05:
### FlutterBluePlus Version

^1.14.7

### Flutter Version

3.7.8

### What OS?

iOS

### OS Version

iOS 14.3

### Ask your question

On iOS, after the Bluetooth device's broadcast signal is first scanned and connected to Bluetooth, the device's broadcast signal is no longer scanned (the Bluetooth device is not connected).

### Logs

```shell
No log
```

barbara012, 2023-09-05:
@wanan76 I also encountered it, and my solution is (a bit tortuous, because flutter_blue_plus is more convenient for subsequent operations):

Use the flutter_reactive_ble library to scan, and then use the scan results to build the scan instance and device instance of flutter_blue_plus.

chipweinberger, 2023-09-05:
@wanan76 when you encounter this problem, is the device discoverable by other BLE scanner apps?

example: (https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner&hl=en_US&gl=US&pli=1)

wanan76, 2023-09-05:
@chipweinberger The device cannot be discovered by other BLE scanner app on iOS; But it works on Android.

wanan76, 2023-09-05:
@barbara012 Thank you. I'll try.

chipweinberger, 2023-09-05:
> The device cannot be discovered by other BLE scanner app on iOS

this typically means your ble device stopped advertising. FBP cannot scan it.

> Android works

what do you mean? what works?

chipweinberger, 2023-09-06:
This is from the readme

---

### Scanning does not find my device

**1. your device uses bluetooth classic, not BLE.**

Headphones, speakers, keyboards, mice, gamepads, & printers all use Bluetooth Classic. 

These devices may be found in System Settings, but they cannot be connected to by FlutterBluePlus. FlutterBluePlus only supports Bluetooth Low Energy.

**2. your device stopped advertising.**

- you might need to reboot your device
- you might need put your device in "discovery mode"
- your phone may have already connected automatically
- another app may have already connected to your device
- another phone may have already connected to your device

Try looking through already connected devices:

```dart
// search already connected devices, including devices
// connected to by other apps
List<BluetoothDevice> system = await FlutterBluePlus.connectedSystemDevices;
for (var d in system) {
    print('${r.device.localName} already connected to! ${r.device.remoteId}');
    if (d.localName == "myBleDevice") {
         await r.connect(); // must connect our app
    }
}
```

**3. your scan filters are wrong.**

- try removing all scan filters
- for `withServices` to work, your device must actively advertise the serviceUUIDs it supports

**4. try a ble scanner app**

Search the App Store for a BLE scanner apps and install it on your phone, and another phone.

Question 1: Is *your phone* (the phone with your flutter app) able to scan it using the 3rd party scanner?

Question 2: Is *another phone* able to scan it using the 3rd party scanner?

chipweinberger, 2023-09-08:
Please reopen a new issue with the details of the above questions, if you still have problems.

closed 2023-09-08 by chipweinberger