chipweinberger/flutter_blue_plus #538: [Bug]: the device turned off，but the disconnect is not called.

zhou5852, 2023-09-05:
### FlutterBluePlus Version

1.14.9

### Flutter Version

3.10.6

### What OS?

iOS

### OS Version

IOS 16

### What happened?

The device turned off ,but the disconnect is not called，I use wireshark and tools to found my device send “
.... ..01 = LLID: Continuation fragment of an L2CAP message, or an Empty PDU (0x1)" between the master and slave,
but I turned off my device, the master send "Empty PDU" ， our API can't be called  because some problem。 May be we should solve this problem 。

[image]

[image]

### Logs

```shell
14915	345.412	Master_0xaf9ab59b	LE 1M	LE LL	192	48442µs	0	0	False	711			L2CAP Fragment
15166	352.282	Slave_0xaf9ab59b	LE 1M	LE LL	4	150µs	1	0	False	852			L2CAP Fragment
15184	353.013	Slave_0xaf9ab59b	LE 1M	LE LL	1	149µs	0	1	False	867			L2CAP Fragment
15266	355.450	Master_0xaf9ab59b	LE 1M	LE LL	1	48670µs	0	0	False	917			L2CAP Fragment
15384	358.718	Slave_0xaf9ab59b	LE 1M	LE LL	96	150µs	0	1	False	984			L2CAP Fragment
15434	360.424	Slave_0xaf9ab59b	LE 1M	LE LL	96	149µs	0	1	False	1019			L2CAP Fragment
15518	362.778	Slave_0xaf9ab59b	LE 1M	LE LL	4	149µs	0	1	False	1069			L2CAP Fragment
15552	363.802	Slave_0xaf9ab59b	LE 1M	LE LL	4	149µs	1	0	False	1090			L2CAP Fragment
15589	364.874	Slave_0xaf9ab59b	LE 1M	LE LL	64	150µs	0	1	False	1112			L2CAP Fragment
```

zhou5852, 2023-09-05:
[image]

chipweinberger, 2023-09-05:
FlutterBluePlus uses the `onConnectioStateChanged` callback from the OS platform to detect disconnection.

do you think this is a bug in iOS?

zhou5852, 2023-09-12:
OK，I'm sorry to late response,Yes!!! it's ok and work well, I test my app, our plugins can get the EmptyPDU not response event and connectionstate changed。

zhou5852, 2023-09-12:
Thanks a lot , I love FlutterBluePlus。

closed 2023-09-12 by zhou5852