chipweinberger/flutter_blue_plus #540: [Bug]: (code: 13) GATT_INVALID_ATTRIBUTE_LENGTH

Nicola-Pozza, 2023-09-06:
### FlutterBluePlus Version

1.14.12

### Flutter Version

3.10.5

### What OS?

Android, iOS

### OS Version

Android 13 - IOS 14.5.1

### What happened?

Try to write 509 byte to pass a file and this error occurred

Error general execute command: FlutterBluePlusException: writeCharacteristic: (code: 13) GATT_INVALID_ATTRIBUTE_LENGTH

### Logs

```shell
[FBP-Android] onMethodCall: writeCharacteristic
[FBP-Android] onCharacteristicChanged: uuid: ed2af5d6-d595-11ea-87d0-0242ac130003
[FBP-Android] onCharacteristicWrite: uuid: ed2af5d6-d595-11ea-87d0-0242ac130003 status: 13
┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
│ #0   DeviceFirmwareUpgradeBloc.upgradeFirmwareNotification.<anonymous closure> (package:xxxx_core/features/firmware_upgrade/device/bloc/device_firmware_upgrade_bloc.dart:209:14)
│ #1   _rootRunUnary (dart:async/zone.dart:1407:47)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ 11:41:42.524 (+0:02:05.759071)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ 🐛 Upgrade notify [0, 0, 24, 0, 0, 0]
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
│ #0   BLECommandsTailBloc._executeCommand (package:xxxx_core/features/bluetooth/bloc/ble_commands_tail_bloc.dart:58:14)
│ #1   <asynchronous suspension>
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ 11:41:42.545 (+0:02:05.779935)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ ⛔ Error general execute command: FlutterBluePlusException: writeCharacteristic: (code: 13) GATT_INVALID_ATTRIBUTE_LENGTH
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
D/[FBP-Android]( 3662): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android]( 3662): [FBP-Android] onCharacteristicWrite: uuid: ed2af5d6-d595-11ea-87d0-0242ac130003 status: 13
┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
│ #0   BLECommandsTailBloc._executeCommand (package:xxxx_core/features/bluetooth/bloc/ble_commands_tail_bloc.dart:58:14)
│ #1   <asynchronous suspension>
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ 11:41:42.651 (+0:02:05.885696)
├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
│ ⛔ Error general execute command: FlutterBluePlusException: writeCharacteristic: (code: 13) GATT_INVALID_ATTRIBUTE_LENGTH
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
D/[FBP-Android]( 3662): [FBP-Android] onCharacteristicChanged: uuid: ed2af5d6-d595-11ea-87d0-0242ac130003
```

chipweinberger, 2023-09-06:
is 1.14.12 the first version with this problem?

how many bytes can you successfully write at most? 

what is your MTU?

chipweinberger, 2023-09-06:
I think your problem is that you need to wait for the requestMtu to finish

Nicola-Pozza, 2023-09-06:
i saw it also in the older version 1.4.0 but only in IOS with the new version also in android
The mtu is 512
Also try with smaller writes

chipweinberger, 2023-09-06:
can you try using writeLarge?

```
import 'dart:math';

// split large writes
// Note: This can only be used for write (with response) to avoid data loss
extension WriteLarge on BluetoothCharacteristic {
  Future<void> writeLarge(List<int> value, int mtu, {int timeout = 15}) async {
    int chunk = mtu-3;
    for (int i = 0; i < value.length; i += chunk) {
      List<int> subvalue = value.sublist(i, min(i + chunk, value.length));
      await write(subvalue, withoutResponse:false, timeout: timeout);
    }
  }
}
```

Nicola-Pozza, 2023-09-06:
i can't use it because i have to wait the answer form the other side

chipweinberger, 2023-09-06:
[image]

chipweinberger, 2023-09-06:
Your firmware does not allow 512 bytes writes.

update your firmware, or write less.

Nicola-Pozza, 2023-09-06:
ok i'll check we the firmware guys.
Thanks

closed 2023-09-06 by Nicola-Pozza

Feptkdie, 2023-09-10:
I have same issue, is this solved?

Feptkdie, 2023-09-10:
How can I split write command?

chipweinberger, 2023-09-10:
the problem is the size of the characteristic 

your device has set a small limit for the size 

you must change your device firmware