chipweinberger/flutter_blue_plus #542: [Bug]: NPE on ANdroid in uuid128

MrCsabaToth, 2023-09-06:
### FlutterBluePlus Version

1.14.17

### Flutter Version

3.14.0-0.2.pre

### What OS?

Android

### OS Version

12

### What happened?

Upgraded to FBP 1.14.17 from 1.14.11 and tried to test it with a BLE device. Upon connection I encountered a NullPointerException.

### Logs

```shell
W/BluetoothGatt(20942): Unhandled exception in callback
W/BluetoothGatt(20942): java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.util.UUID.toString()' on a null object reference
W/BluetoothGatt(20942): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.uuid128(FlutterBluePlusPlugin.java:114)
W/BluetoothGatt(20942): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmBluetoothCharacteristic(FlutterBluePlusPlugin.java:1837)
W/BluetoothGatt(20942): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.bmBluetoothService(FlutterBluePlusPlugin.java:1804)
W/BluetoothGatt(20942): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$4.onServicesDiscovered(FlutterBluePlusPlugin.java:1545)
W/BluetoothGatt(20942): 	at android.bluetooth.BluetoothGatt$1$5.run(BluetoothGatt.java:347)
W/BluetoothGatt(20942): 	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:823)
W/BluetoothGatt(20942): 	at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:47)
W/BluetoothGatt(20942): 	at android.bluetooth.BluetoothGatt$1.onSearchComplete(BluetoothGatt.java:342)
W/BluetoothGatt(20942): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:241)
W/BluetoothGatt(20942): 	at android.os.Binder.execTransactInternal(Binder.java:1226)
W/BluetoothGatt(20942): 	at android.os.Binder.execTransact(Binder.java:1163)
```
```

MrCsabaToth, 2023-09-06:
Hmmm, kinda weird, because line 1837 which calls the `uuid128` has a null check: `map.put("secondary_service_uuid", pair.secondary != null ? uuid128(pair.secondary) : null);`. So maybe the stack line numbering could be off somehow? Even though the line 114 matches, but that's at the top of the file.

MrCsabaToth, 2023-09-06:
The crash happens at the `discoverServices` call and not the `connect`. All devices I tried are affected.

chipweinberger, 2023-09-06:
fixed in 1.14.18

closed 2023-09-06 by chipweinberger

MrCsabaToth, 2023-09-07:
Looks like the crash is cured according to my tests.