chipweinberger/flutter_blue_plus #543: [Bug]: infinte loop still exists in v1.14.20 when hot restart

project2021-code, 2023-09-06:
### FlutterBluePlus Version

1.14.20

### Flutter Version

3.13.2

### What OS?

Android

### OS Version

Android 8

### What happened?

Cannot open Bluetooth after Hot Restart when connected to BLE GATT device

### Logs

```shell
Performing hot restart...
D/[FBP-Android](18716): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18716): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] onReadRemoteRssi: rssi: -73 status: 0
D/[FBP-Android](18716): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18716): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] onReadRemoteRssi: rssi: -67 status: 0
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a2
D/[FBP-Android](18716): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18716): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] onReadRemoteRssi: rssi: -71 status: 0
D/[FBP-Android](18716): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18716): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] onReadRemoteRssi: rssi: -55 status: 0
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android](18716): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18716): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] onReadRemoteRssi: rssi: -51 status: 0
D/[FBP-Android](18716): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(18716): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] onReadRemoteRssi: rssi: -70 status: 0
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
Restarted application in 4,172ms.
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/ViewRootImpl@2938c5[MainActivity](18716): Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x1 surface={valid=true 493594169344} changed=false
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/ViewRootImpl@2938c5[MainActivity](18716): ViewPostIme pointer 0
D/ViewRootImpl@2938c5[MainActivity](18716): ViewPostIme pointer 1
D/[FBP-Android](18716): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](18716): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](18716): [FBP-Android] closeAllConnections
D/[FBP-Android](18716): [FBP-Android] calling disconnect() on device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] calling gatt.close() on device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): close()
D/BluetoothGatt(18716): unregisterApp() - mClientIf=6
D/BluetoothGatt(18716): onClientConnectionState() - status=0 clientIf=6 device=30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] connectedPeripherals: 1
D/[FBP-Android](18716): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](18716): [FBP-Android] closeAllConnections
D/[FBP-Android](18716): [FBP-Android] calling disconnect() on device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] calling gatt.close() on device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): close()
D/BluetoothGatt(18716): unregisterApp() - mClientIf=0
D/[FBP-Android](18716): [FBP-Android] connectedPeripherals: 1
D/[FBP-Android](18716): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](18716): [FBP-Android] closeAllConnections
D/[FBP-Android](18716): [FBP-Android] calling disconnect() on device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] calling gatt.close() on device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): close()
D/BluetoothGatt(18716): unregisterApp() - mClientIf=0
D/[FBP-Android](18716): [FBP-Android] connectedPeripherals: 1
D/[FBP-Android](18716): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](18716): [FBP-Android] closeAllConnections
D/[FBP-Android](18716): [FBP-Android] calling disconnect() on device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] calling gatt.close() on device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): close()
D/BluetoothGatt(18716): unregisterApp() - mClientIf=0
D/[FBP-Android](18716): [FBP-Android] connectedPeripherals: 1
D/[FBP-Android](18716): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](18716): [FBP-Android] closeAllConnections
D/[FBP-Android](18716): [FBP-Android] calling disconnect() on device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] calling gatt.close() on device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): close()
D/BluetoothGatt(18716): unregisterApp() - mClientIf=0
D/[FBP-Android](18716): [FBP-Android] connectedPeripherals: 1
D/[FBP-Android](18716): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](18716): [FBP-Android] closeAllConnections
D/[FBP-Android](18716): [FBP-Android] calling disconnect() on device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] calling gatt.close() on device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): close()
D/BluetoothGatt(18716): unregisterApp() - mClientIf=0
D/[FBP-Android](18716): [FBP-Android] connectedPeripherals: 1
D/[FBP-Android](18716): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](18716): [FBP-Android] closeAllConnections
D/[FBP-Android](18716): [FBP-Android] calling disconnect() on device: 30:AE:A4:28:2C:CE
D/[FBP-Android](18716): [FBP-Android] calling gatt.close() on device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(18716): close()
D/BluetoothGatt(18716): unregisterApp() - mClientIf=0
D/[FBP-Android](18716): [FBP-Android] connectedPeripherals: 1
Lost connection to device.
```

chipweinberger, 2023-09-06:
Can you try master branch? I've pushed a fix 0fad3440df0d5fdbcc816296ba2565394dd362c8

chipweinberger, 2023-09-06:
should be fixed in 1.14.22

closed 2023-09-06 by chipweinberger