chipweinberger/flutter_blue_plus #544: [Bug]: Cannot connect to device after connected once and disconenct

project2021-code, 2023-09-06:
### FlutterBluePlus Version

1.14.22

### Flutter Version

3.13.2

### What OS?

Android

### OS Version

Android 8

### What happened?

After connected to device and disconnect it, I cannot connect to device.

Tried turn off Bluetooth and Turn on, no luck

Hot Restart can reset the state, I can connect the device again but the issue still exists after disconnect the device.

### Logs

```shell
D/[FBP-Android](22237): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](22237): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android](22237): [FBP-Android] disconnectAllDevices
D/[FBP-Android](22237): [FBP-Android] connectedPeripherals: 0
D/[FBP-Android](22237): [FBP-Android] HotRestart: complete
D/[FBP-Android](22237): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](22237): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](22237): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(22237): getConnectedDevices
D/[FBP-Android](22237): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(22237): STATE_ON
D/BluetoothAdapter(22237): STATE_ON
D/BluetoothLeScanner(22237): Start Scan with callback
D/BluetoothLeScanner(22237): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android](22237): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(22237): STATE_ON
D/BluetoothAdapter(22237): STATE_ON
D/BluetoothLeScanner(22237): Stop Scan with callback
D/[FBP-Android](22237): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(22237): connect() - device: 30:AE:A4:28:2C:CE, auto: false
D/BluetoothAdapter(22237): isSecureModeEnabled
D/BluetoothGatt(22237): registerApp()
D/BluetoothGatt(22237): registerApp() - UUID=29d07d0f-7f28-4fef-80b0-04e7e130a6d5
D/BluetoothGatt(22237): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt(22237): onClientConnectionState() - status=0 clientIf=6 device=30:AE:A4:28:2C:CE
D/[FBP-Android](22237): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](22237): [FBP-Android] onMethodCall: connect
D/[FBP-Android](22237): [FBP-Android] already connected
D/[FBP-Android](22237): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt(22237): configureMTU() - device: 30:AE:A4:28:2C:CE mtu: 185
D/BluetoothGatt(22237): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(22237): onConfigureMTU() - Device=30:AE:A4:28:2C:CE mtu=185 status=0
D/[FBP-Android](22237): [FBP-Android] onMtuChanged: mtu: 185 status: 0
D/[FBP-Android](22237): [FBP-Android] onMethodCall: getAdapterState
D/BluetoothGatt(22237): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=39 latency=0 timeout=500 status=0
D/[FBP-Android](22237): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(22237): discoverServices() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(22237): onSearchComplete() = Device=30:AE:A4:28:2C:CE Status=0
D/[FBP-Android](22237): [FBP-Android] onServicesDiscovered: count: 4 status: 0
D/[FBP-Android](22237): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(22237): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a2 enable: true
D/[FBP-Android](22237): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](22237): [FBP-Android] onMethodCall: setNotification
[... 128 lines cut ...]
D/BluetoothAdapter(22237): STATE_ON
D/BluetoothAdapter(22237): STATE_ON
D/BluetoothLeScanner(22237): Stop Scan with callback
D/[FBP-Android](22237): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(22237): connect() - device: 30:AE:A4:28:2C:CE, auto: false
D/BluetoothAdapter(22237): isSecureModeEnabled
D/BluetoothGatt(22237): registerApp()
D/BluetoothGatt(22237): registerApp() - UUID=78f26c0a-d7b3-49e8-bea3-e6b1ad203f21
D/BluetoothGatt(22237): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt(22237): onClientConnectionState() - status=0 clientIf=6 device=30:AE:A4:28:2C:CE
D/[FBP-Android](22237): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](22237): [FBP-Android] onMethodCall: connect
D/[FBP-Android](22237): [FBP-Android] already connected
D/[FBP-Android](22237): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt(22237): configureMTU() - device: 30:AE:A4:28:2C:CE mtu: 185
D/BluetoothGatt(22237): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(22237): onConfigureMTU() - Device=30:AE:A4:28:2C:CE mtu=185 status=0
D/[FBP-Android](22237): [FBP-Android] onMtuChanged: mtu: 185 status: 0
D/[FBP-Android](22237): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](22237): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(22237): discoverServices() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(22237): onSearchComplete() = Device=30:AE:A4:28:2C:CE Status=0
D/[FBP-Android](22237): [FBP-Android] onServicesDiscovered: count: 4 status: 0
D/BluetoothGatt(22237): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=39 latency=0 timeout=500 status=0
```

chipweinberger, 2023-09-06:
Your logs show that you connect without issues.

```
D/[FBP-Android](22237): [FBP-Android] onMethodCall: connect
...
D/[FBP-Android](22237): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
...
D/[FBP-Android](22237): [FBP-Android] onMethodCall: connect
D/[FBP-Android](22237): [FBP-Android] already connected
```

chipweinberger, 2023-09-06:
Because hot restart can fix it, the problem is most likely in your dart code. Hot Restart restarts the dart vm.

project2021-code, 2023-09-07:
I changed the title, the problem should be on **characteristic.lastValueStream** or **characteristic.onValueReceived**.

In my code, when the device disconnected, I didn't set the notify to false before disconnection. In v1.14.18 or before, the problem does not exist.

Also, I find that if I call characteristic.setNotifyValue(false) after disconnect the device, I will not able to connect to device.

chipweinberger, 2023-09-07:
I don't understand. What is the problem?

project2021-code, 2023-09-07:
1. connect device
2. search service
3. search GATT characteristic
4. subscribe GATT characteristic
5. get data from characteristic.lastValueStream or characteristic.onValueReceived
6. disconnect device
7. connect device
8. search service
9. search GATT characteristic
10. subscribe GATT characteristic
11. **cannot get data from characteristic.lastValueStream or characteristic.onValueReceived**

project2021-code, 2023-09-07:
My workaround is set **characteristic.setNotifyValue(true)** to the characteristic.

If I do not set **characteristic.setNotifyValue(false)** before disconnect the device, the **characteristic.isNotifying remain true**.

```
await characteristic.setNotifyValue(true);
// if (!characteristic.isNotifying) {
//   await characteristic.setNotifyValue(!characteristic.isNotifying);
// }
```

chipweinberger, 2023-09-07:
oh ok, i know the problem now. Will fix.

project2021-code, 2023-09-07:
Not sure if I should create a new issue. I find another another problem:

If characteristic.setNotifyValue(false) is called after the disconnect the device, I cannot connect to device.
```
D/[FBP-Android](27496): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android](27496): [FBP-Android] disconnectAllDevices
D/[FBP-Android](27496): [FBP-Android] connectedPeripherals: 0
D/[FBP-Android](27496): [FBP-Android] HotRestart: complete
D/[FBP-Android](27496): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](27496): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](27496): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager(27496): getConnectedDevices
D/[FBP-Android](27496): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(27496): STATE_ON
D/BluetoothAdapter(27496): STATE_ON
D/BluetoothLeScanner(27496): Start Scan with callback
D/BluetoothLeScanner(27496): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(27496): STATE_ON
D/BluetoothAdapter(27496): STATE_ON
D/BluetoothAdapter(27496): STATE_ON
D/BluetoothLeScanner(27496): Stop Scan with callback
D/[FBP-Android](27496): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(27496): connect() - device: 30:AE:A4:28:2C:CE, auto: false
D/BluetoothAdapter(27496): isSecureModeEnabled
D/BluetoothGatt(27496): registerApp()
D/BluetoothGatt(27496): registerApp() - UUID=cfac8a42-9992-42e9-a3cd-a966614d8d6f
D/BluetoothGatt(27496): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt(27496): onClientConnectionState() - status=0 clientIf=6 device=30:AE:A4:28:2C:CE
D/[FBP-Android](27496): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](27496): [FBP-Android] onMethodCall: connect
D/[FBP-Android](27496): [FBP-Android] already connected
D/[FBP-Android](27496): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt(27496): configureMTU() - device: 30:AE:A4:28:2C:CE mtu: 185
D/BluetoothGatt(27496): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(27496): onConfigureMTU() - Device=30:AE:A4:28:2C:CE mtu=185 status=0
D/[FBP-Android](27496): [FBP-Android] onMtuChanged: mtu: 185 status: 0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android](27496): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(27496): discoverServices() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt(27496): onSearchComplete() = Device=30:AE:A4:28:2C:CE Status=0
D/[FBP-Android](27496): [FBP-Android] onServicesDiscovered: count: 4 status: 0
D/BluetoothGatt(27496): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=39 latency=0 timeout=500 status=0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(27496): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a2 enable: true
D/[FBP-Android](27496): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(27496): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1 enable: true
D/[FBP-Android](27496): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(27496): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4 enable: true
D/[FBP-Android](27496): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android](27496): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](27496): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(27496): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](27496): [FBP-Android] onReadRemoteRssi: rssi: -53 status: 0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(27496): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](27496): [FBP-Android] onReadRemoteRssi: rssi: -50 status: 0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(27496): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](27496): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android](27496): [FBP-Android] onReadRemoteRssi: rssi: -47 status: 0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(27496): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](27496): [FBP-Android] onReadRemoteRssi: rssi: -49 status: 0
D/[FBP-Android](27496): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt(27496): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android](27496): [FBP-Android] onReadRemoteRssi: rssi: -50 status: 0
D/[FBP-Android](27496): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
[... 57 lines cut ...]
I/flutter (27496): BT State in Off Screen: BluetoothAdapterState.turningOn
D/BluetoothAdapter(27496): onBluetoothStateChange: up=true
D/BluetoothAdapter(27496): onBluetoothStateChange: Bluetooth is on
D/[FBP-Android](27496): [FBP-Android] OnAdapterStateChanged: on
I/flutter (27496): BT State in Off Screen: BluetoothAdapterState.unknown
D/[FBP-Android](27496): [FBP-Android] OnAdapterStateChanged: turningOff
D/[FBP-Android](27496): [FBP-Android] disconnectAllDevices
[... cut ...]
D/BluetoothAdapter(27496): There are no active google scan apps, stop scan
D/BluetoothLeScanner(27496): stopAllScan standalone boolean is value is = false
D/BluetoothLeScanner(27496): Exting stopping all callback scans
D/BluetoothLeScanner(27496): Exting stopping all pending intent scans
D/[FBP-Android](27496): [FBP-Android] OnAdapterStateChanged: turningOn
D/BluetoothAdapter(27496): onBluetoothStateChange: up=true
D/BluetoothAdapter(27496): onBluetoothStateChange: Bluetooth is on
D/[FBP-Android](27496): [FBP-Android] OnAdapterStateChanged: on
```

chipweinberger, 2023-09-07:
fixed in 1.14.23

open a new issue if you still have problems

closed 2023-09-07 by chipweinberger