chipweinberger/flutter_blue_plus #545: [Bug]: flutter_blue_plus crashed if characteristic.setNotifyValue(false) called after device disconnect

project2021-code, 2023-09-07:
### FlutterBluePlus Version

1.14.23

### Flutter Version

3.13.2

### What OS?

Android

### OS Version

Android 8

### What happened?

If characteristic.setNotifyValue(false) called after disconnect the device, I cannot connect to device.

Bluetooth Scan cannot be called.
Bluetooth State become Unknown.

### Logs

```shell
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: flutterHotRestart
D/[FBP-Android]( 1409): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android]( 1409): [FBP-Android] disconnectAllDevices
D/[FBP-Android]( 1409): [FBP-Android] connectedPeripherals: 0
D/[FBP-Android]( 1409): [FBP-Android] HotRestart: complete
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/BluetoothManager( 1409): getConnectedDevices
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter( 1409): STATE_ON
D/BluetoothAdapter( 1409): STATE_ON
D/BluetoothLeScanner( 1409): Start Scan with callback
D/BluetoothLeScanner( 1409): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 1409): STATE_ON
D/BluetoothAdapter( 1409): STATE_ON
D/BluetoothAdapter( 1409): STATE_ON
D/BluetoothLeScanner( 1409): Stop Scan with callback
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: connect
D/BluetoothGatt( 1409): connect() - device: 30:AE:A4:28:2C:CE, auto: false
D/BluetoothAdapter( 1409): isSecureModeEnabled
D/BluetoothGatt( 1409): registerApp()
D/BluetoothGatt( 1409): registerApp() - UUID=72b73b51-fce3-4e19-bdf5-1d6449e93ece
D/BluetoothGatt( 1409): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt( 1409): onClientConnectionState() - status=0 clientIf=6 device=30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: connect
D/[FBP-Android]( 1409): [FBP-Android] already connected
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt( 1409): configureMTU() - device: 30:AE:A4:28:2C:CE mtu: 185
D/BluetoothGatt( 1409): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 1409): onConfigureMTU() - Device=30:AE:A4:28:2C:CE mtu=185 status=0
D/[FBP-Android]( 1409): [FBP-Android] onMtuChanged: mtu: 185 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt( 1409): discoverServices() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt( 1409): onSearchComplete() = Device=30:AE:A4:28:2C:CE Status=0
D/[FBP-Android]( 1409): [FBP-Android] onServicesDiscovered: count: 4 status: 0
D/BluetoothGatt( 1409): onConnectionUpdated() - Device=30:AE:A4:28:2C:CE interval=39 latency=0 timeout=500 status=0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt( 1409): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a2 enable: true
D/[FBP-Android]( 1409): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt( 1409): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1 enable: true
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android]( 1409): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt( 1409): setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4 enable: true
D/[FBP-Android]( 1409): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -56 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -56 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -57 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -60 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -58 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -59 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -61 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -57 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a1
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android]( 1409): [FBP-Android] onCharacteristicChanged: uuid: beb5483e-36e1-4688-b7f5-ea07361b26a4
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: readRssi
D/BluetoothGatt( 1409): readRssi() - device: 30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onReadRemoteRssi: rssi: -60 status: 0
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: disconnect
D/BluetoothGatt( 1409): cancelOpen() - device: 30:AE:A4:28:2C:CE
D/BluetoothGatt( 1409): onClientConnectionState() - status=0 clientIf=6 device=30:AE:A4:28:2C:CE
D/[FBP-Android]( 1409): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt( 1409): close()
D/BluetoothGatt( 1409): unregisterApp() - mClientIf=6
D/[FBP-Android]( 1409): [FBP-Android] onMethodCall: setNotification
E/flutter ( 1409): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(setNotification, device is disconnected, null, null)
E/flutter ( 1409): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter ( 1409): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter ( 1409): <asynchronous suspension>
E/flutter ( 1409): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:339:19)
E/flutter ( 1409): <asynchronous suspension>
E/flutter ( 1409): #3      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:215:5)
E/flutter ( 1409): <asynchronous suspension>
E/flutter ( 1409): 
I/flutter ( 1409): BT State in Off Screen: BluetoothAdapterState.unknown
```

chipweinberger, 2023-09-07:
this is expected.

setNotifyValue cannot work when the device is disconnected, so it throws an error.

You should either handle the error, or not call setNotifyValue when the device is disconnected.

chipweinberger, 2023-09-07:
thanks for filing these issues!

closed 2023-09-07 by chipweinberger