chipweinberger/flutter_blue_plus #546: [Bug]: LateInitializationError: Field 'characteristic' has not been initialized. & PlatformException(discoverServices, device is disconnected, null, null)

widzinskaj, 2023-09-07:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.13.2

### What OS?

Android

### OS Version

Android 13

### What happened?

I had a version of  working application in Android Studio and I made .apk file. I don't know what changed, but suddenly, when I returned to work over the more developed version (which also worked!) I got the problem which didn't occured earlier.

As you can see from logs, I had a problem with discoverServices, but it doesn't stop the app. The problem occurs when I push the button and I obtain LateInitializationError. It appears when I push the button and it writes  to BluetoothCharacteristic a message to turn off the device. I don't know what happened, because the more developed version worked before making an .apk file.

It's my first more advanced application and I don't know where is the problem - Flutter Blue Plus library, configuration files...I really appreciate any help.

### Logs

```shell
Launching lib\main.dart on moto e13 in debug mode...
Running Gradle task 'assembleDebug'...
√  Built build\app\outputs\flutter-apk\app-debug.apk.
Debug service listening on ws://127.0.0.1:52746/5DAd2jR_dsQ=/ws
Syncing files to device moto e13...
D/[FBP-Android](15441): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](15441): [FBP-Android] initializing BluetoothAdapter
D/[FBP-Android](15441): [FBP-Android] onMethodCall: getAdapterState
I/mali_gralloc(15441): register: id=0x25f0000032d, importpid=-1
I/mali_gralloc(15441): register: id=0x25f0000032e, importpid=-1
W/Parcel  (15441): Expecting binder but got null!
W/ziparchive(15441): Unable to open '/system_ext/framework/unipnp-framework.dm': No such file or directory
W/ziparchive(15441): Unable to open '/system_ext/framework/unipnp-framework.dm': No such file or directory
I/mali_gralloc(15441): register: id=0x25f0000032f, importpid=-1
D/[FBP-Android](15441): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(15441): isLeEnabled(): ON
D/BluetoothLeScanner(15441): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
D/[FBP-Android](15441): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(15441): connect() - device: 60:8A:10:50:FE:56, auto: false
D/BluetoothGatt(15441): registerApp()
D/BluetoothGatt(15441): registerApp() - UUID=4f608116-b935-4219-8b07-6e9ccdc5a7b4
D/BluetoothGatt(15441): onClientRegistered() - status=0 clientIf=5
D/[FBP-Android](15441): [FBP-Android] onMethodCall: discoverServices
E/flutter (15441): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(discoverServices, device is disconnected, null, null)
E/flutter (15441): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (15441): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (15441): <asynchronous suspension>
E/flutter (15441): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:302:19)
E/flutter (15441): <asynchronous suspension>
E/flutter (15441): #3      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:172:7)
E/flutter (15441): <asynchronous suspension>
E/flutter (15441): 
I/mali_gralloc(15441): register: id=0x25f00000330, importpid=-1
D/BluetoothGatt(15441): onClientConnectionState() - status=0 clientIf=5 device=60:8A:10:50:FE:56
D/[FBP-Android](15441): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](15441): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(15441): discoverServices() - device: 60:8A:10:50:FE:56
D/BluetoothGatt(15441): onSearchComplete() = Device=60:8A:10:50:FE:56 Status=0
D/[FBP-Android](15441): [FBP-Android] onServicesDiscovered: count: 5 status: 0
D/[FBP-Android](15441): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(15441): isLeEnabled(): ON
D/[FBP-Android](15441): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt(15441): setCharacteristicNotification() - uuid: 49535343-1e4d-4bd9-ba61-23c647249616 enable: true
[... 105 lines cut ...]
            characteristic = c;
            sendRepeatableTaskOldBMS(c);
          } else if (c.uuid == newbmsrx) {
            //read new BMS
            characteristicToNotify = c;
            await c.setNotifyValue(!c.isNotifying);
            c.value.listen((event) {
              temp = temp + event;
              widget.device.battery.processReceivedDataNewBMS(temp);
              if (mounted) {
                setState(() {
                  temp;
                });
              }
            });
          } else if (c.uuid == newbmstx) {
            //write new BMS
            characteristic = c;
            sendRepeatableTaskNewBMS(c);
          }
        }
      }
    });
  }
```

chipweinberger, 2023-09-07:
seems like bugs in your own code. try asking chatgpt

chipweinberger, 2023-09-07:
closing since these are not FBP bugs

closed 2023-09-07 by chipweinberger