chipweinberger/flutter_blue_plus #547: [Question]:

brain9520, 2023-09-08:
### FlutterBluePlus Version

1.14.23

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

13

### Ask your question

Is there a parameter to set not to scan devices repeatedly? thank you.

### Logs

```shell
When I use the Android device to start scanning, it will display as follows:
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EC100200
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540
I/flutter (18948): ble_device_name: EA02001022540

When I scan with an Apple device as follows:
flutter: ble_device_name: EA02001022540
flutter: ble_device_name: midea
flutter: ble_device_name: midea
flutter: ble_device_name: EC100200
flutter: ble_device_name: KeepB3-31
flutter: ble_device_name: HUAWEI Band 7-D5C
flutter: ble_device_name: MITV-2E531
flutter: ble_device_name: HUAWEI Band 7-BFF
flutter: ble_device_name: MITV-900BB
```

chipweinberger, 2023-09-08:
what does your code look like?

`allowDuplicates` is the option you can control this, usually.

brain9520, 2023-09-08:
> what does your code look like?
> 
> `allowDuplicates` is the option you can control this, usually. But depends on your code.
> 
> this could also be a bug in android itself

FlutterBluePlus.scan(timeout: const Duration(seconds: 4),allowDuplicates: false).listen ， I have set this parameter to no avail

brain9520, 2023-09-08:
> use this code, which is stronger at detecting duplicates
> 
> ```
> // Setup Listener for scan results
> // device not found? see "Common Problems" in the README
> var subscription = FlutterBluePlus.scanResults.listen((results) {
>     for (ScanResult r in results) {
>         print('${r.device.localName} found! rssi: ${r.rssi}');
>     }
> });
> 
> // Start scanning
> FlutterBluePlus.startScan(timeout: Duration(seconds: 4));
> 
> // Stop scanning
> await FlutterBluePlus.stopScan();
> ```

Thank you so much for your selfless help.

chipweinberger, 2023-09-08:
It appears there actually is a bug in FBP, when using scan. I'll push a fix

chipweinberger, 2023-09-08:
this is now fixed in 1.15.1

closed 2023-09-08 by chipweinberger

brain9520, 2023-09-08:
> this is now fixed in 1.15.1

Hello, I have upgraded the version to 1.15.1, and the encoding of my search device is as follows:
`   FlutterBluePlus.scanResults.listen((results) {
      for (ScanResult r in results) {
        debugPrint('${r.device.localName} found! rssi: ${r.rssi}');
      }
    });
    // start
    await FlutterBluePlus.startScan(
      oneByOne: true,
      timeout: const Duration(seconds: 4),
      removeIfGone: const Duration(seconds: 1), 
    );`
When I run the program, the console still displays many duplicate devices, as follows:

`
2
I/flutter (17410): midea found! rssi: -81
I/flutter (17410): EC100200 found! rssi: -76
I/flutter (17410): EA222228 found! rssi: -59
I/flutter (17410): LE-reserved_C found! rssi: -101
I/flutter (17410): midea found! rssi: -79
I/flutter (17410): EC100200 found! rssi: -49
I/flutter (17410): EA222228 found! rssi: -59
I/flutter (17410): midea found! rssi: -83
I/flutter (17410): midea found! rssi: -87
I/flutter (17410): midea found! rssi: -79
I/flutter (17410): EC100200 found! rssi: -55
I/flutter (17410): EA222228 found! rssi: -59
I/flutter (17410): midea found! rssi: -83
I/flutter (17410): midea found! rssi: -87
I/flutter (17410): EA222228 found! rssi: -60
I/flutter (17410): midea found! rssi: -86
I/flutter (17410): midea found! rssi: -86
I/flutter (17410): midea found! rssi: -82
I/flutter (17410): midea found! rssi: -86
I/flutter (17410): midea found! rssi: -87
I/flutter (17410): EC100200 found! rssi: -49
I/flutter (17410): EA222228 found! rssi: -59
I/flutter (17410): midea found! rssi: -87
I/flutter (17410): EC100200 found! rssi: -53
I/flutter (17410): midea found! rssi: -83
I/flutter (17410): EC100200 found! rssi: -65
I/flutter (17410): EA222228 found! rssi: -61
I/flutter (17410): midea found! rssi: -86
I/flutter (17410): midea found! rssi: -79
I/flutter (17410): Mi Smart Band 4 found! rssi: -79
I/flutter (17410): EC100200 found! rssi: -50
I/flutter (17410): EA222228 found! rssi: -59
I/flutter (17410): midea found! rssi: -86
`

When I was doing Android native development before, I used a third-party library. There were four methods in the scan callback of that library, among which
` @Override
      public void onScanning(BleDevice bleDevice) {

      }`
Duplicate device names will not be returned in the callback，
`override fun onLeScan(bleDevice: BleDevice?) {
                super.onLeScan(bleDevice) 
            }`
Duplicate device names will be returned in the callback.
But I don't have the ability to port the implementation over there to flutter_blue_plus.

brain9520, 2023-09-08:
> > this is now fixed in 1.15.1
> 
> Hello, I have upgraded the version to 1.15.1, and the encoding of my search device is as follows:
> 
> 
`
 FlutterBluePlus.scanResults.listen((results) { for (ScanResult r in results) { debugPrint('${r.device.localName} found! rssi: ${r.rssi}'); } }); // start await FlutterBluePlus.startScan( oneByOne: true, timeout: const Duration(seconds: 4), removeIfGone: const Duration(seconds: 1), );
`
> 
> When I run the program, the console still displays many duplicate devices, as follows:
> 
> `I/flutter (17410): midea found! rssi: -81 I/flutter (17410): EC100200 found! rssi: -76 I/flutter (17410): EA222228 found! rssi: -59 I/flutter (17410): LE-reserved_C found! rssi: -101 I/flutter (17410): midea found! rssi: -79 I/flutter (17410): EC100200 found! rssi: -49 I/flutter (17410): EA222228 found! rssi: -59 I/flutter (17410): midea found! rssi: -83 I/flutter (17410): midea found! rssi: -87 I/flutter (17410): midea found! rssi: -79 I/flutter (17410): EC100200 found! rssi: -55 I/flutter (17410): EA222228 found! rssi: -59 I/flutter (17410): midea found! rssi: -83 I/flutter (17410): midea found! rssi: -87 I/flutter (17410): EA222228 found! rssi: -60 I/flutter (17410): midea found! rssi: -86 I/flutter (17410): midea found! rssi: -86 I/flutter (17410): midea found! rssi: -82 I/flutter (17410): midea found! rssi: -86 I/flutter (17410): midea found! rssi: -87 I/flutter (17410): EC100200 found! rssi: -49 I/flutter (17410): EA222228 found! rssi: -59 I/flutter (17410): midea found! rssi: -87 I/flutter (17410): EC100200 found! rssi: -53 I/flutter (17410): midea found! rssi: -83 I/flutter (17410): EC100200 found! rssi: -65 I/flutter (17410): EA222228 found! rssi: -61 I/flutter (17410): midea found! rssi: -86 I/flutter (17410): midea found! rssi: -79 I/flutter (17410): Mi Smart Band 4 found! rssi: -79 I/flutter (17410): EC100200 found! rssi: -50 I/flutter (17410): EA222228 found! rssi: -59 I/flutter (17410): midea found! rssi: -86`
> 
> When I was doing Android native development before, I used a third-party library. There were four methods in the scan callback of that library, among which ` @OverRide public void onScanning(BleDevice bleDevice) {
> 
> ```
>   }
> ```
> 
> `Duplicate device names will not be returned in the callback，` override fun onLeScan(bleDevice: BleDevice?) { super.onLeScan(bleDevice) } ` Duplicate device names will be returned in the callback. But I don't have the ability to port the implementation over there to flutter_blue_plus.

chipweinberger, 2023-09-08:
> When I run the program, the console still displays many duplicate devices

This is expected because you are using `oneByOne: true`

chipweinberger, 2023-09-08:
I've updated the comment. hopefully more clear

  ///   - [oneByOne] if true, we will stream every advertistment one by one, including duplicates.
  ///    If false, we deduplicate the advertisements, and return a list of devices.

brain9520, 2023-09-08:
I tried the parameters you mentioned, if I set oneByOne: false，Scanning will not stop, even if I set a timeout, it will continue to scan.

brain9520, 2023-09-08:
> I've updated the comment. hopefully more clear
> 
> ```
>   ///   - [oneByOne] if true, we will stream every advertistment one by one, including duplicates.
>   ///    If false, we deduplicate the advertisements, and return a list of devices.
> ```

my codes：

     FlutterBluePlus.scanResults.listen((results) {
      for (ScanResult r in results) {
        // if (r.device.localName.isEmpty) return;q
        debugPrint(
            '${r.device.localName} found! rssi: ${r.rssi}-- time: ${DateTime.now()}');
      }
    });
    
    await FlutterBluePlus.startScan(
      // oneByOne: false,
      timeout: const Duration(seconds: 4),
      removeIfGone: const Duration(seconds: 1), 
    );

brain9520, 2023-09-08:
There is another problem here, if I uncomment the // if (r.device.localName.isEmpty) return; uncommented in the code, I find that the name of any device will not be printed in the console, but in fact I am nearby Many Bluetooth devices.

chipweinberger, 2023-09-08:
not sure, lmk if you figure it out

brain9520, 2023-09-08:
> not sure, lmk if you figure it out

[error_info.pdf](https://github.com/boskokg/flutter_blue_plus/files/12557691/error_info.pdf)

chipweinberger, 2023-09-08:
try this

late List<ScanResult> last;

FlutterBluePlus.scanResults.listen((results) {last = results;});

await FlutterBluePlus.startScan(timeout: Duration(seconds: 10);

await Future.delayed(Duration(seconds:11));

for (ScanResult r in last) {
    print('${r.device.localName} found!  ${r.device.remoteId}  rssi: ${r.rssi}');
}

brain9520, 2023-09-08:
> 

    FlutterBluePlus.scanResults.listen((results) {
       // This callback method is called every time the scanned array changes, right? I think it would be more convenient to add some parameter controllers as a screening mechanism, for example, the scanned device has a name and contains a certain prefix. Will adding these parameters make this monitoring method less likely to be called?
    });

Then, I really appreciate you keep replying to my questions, please excuse my bad technique。 thank you very much!

brain9520, 2023-09-08:
> > 
> 
@chipweinberger 
> ```
> FlutterBluePlus.scanResults.listen((results) {
>    // This callback method is called every time the scanned array changes, right? I think it would be more convenient to add some parameter [...]

chipweinberger, 2023-09-08:
> This callback method is called every time the scanned array changes, right? 

yes.

simple code to remove duplicates.

```
Set<DeviceIdentifier> printed = {};
var subscription = FlutterBluePlus.scanResults.listen((results) {
    for (ScanResult r in results) {
        if (printed.contains(r.device.remoteId) == false) {
            print('${r.device.remoteId}: "${r.device.localName}" found! rssi: ${r.rssi}');
            printed.add(r.device.remoteId);
        }
    }
});
```

chipweinberger, 2023-09-08:
For example:

```
  FlutterBluePlus.scanResults
      .map((list) => list.first)
      .where((r) => r.device.localName == "example")
      .where((r) => got.contains(r.device.remoteId) == false)
      .listen((r) {
    print('${r.device.remoteId}: "${r.device.localName}" found! rssi: ${r.rssi}');
  });

await FlutterBluePlus.startScan(timeout: Duration(seconds: 10), oneByOne:true);
```