chipweinberger/flutter_blue_plus #548: [Bug]:

brain9520, 2023-09-08:
### FlutterBluePlus Version

1.14.23

### Flutter Version

3.10.6

### What OS?

iOS

### OS Version

14.0.1

### What happened?

Here's how I code it when I click a button to start scanning for bluetooth devices:
if (await FlutterBluePlus.isAvailable == false) {
      EasyLoading.showError("phone_not_support_ble".tr);
      return;
    }
 
    var isOpen = await FlutterBluePlus.adapterState.first;
    if (isOpen != BluetoothAdapterState.on) {
      if (Platform.isAndroid) {
        await FlutterBluePlus.turnOn();
      }
      
      if (Platform.isIOS) {
        EasyLoading.showError('phone_open_ble_first'.tr);
      }
      return;
    }

This encoding works normally on Android devices, but when Bluetooth is turned on when the button is clicked for the first time on an iOS device, an error message will pop up, and then the second click will start scanning normally.

### Logs

```shell
Clicking the button for the first time will definitely enter this judgment 
if (Platform.isIOS) {
        EasyLoading.showError('phone_open_ble_first'.tr);
 }
```

chipweinberger, 2023-09-08:
this is a bug in your own code. cannot help you.

chipweinberger, 2023-09-08:
// wait bluetooth to be on & print states
// note: for iOS the initial state is typically BluetoothAdapterState.unknown
// note: if you have permissions issues you will get stuck at BluetoothAdapterState.unauthorized
await FlutterBluePlus.adapterState
 .map((s){print(s);return s;})
 .where((s) => s == BluetoothAdapterState.on)
 .first;

closed 2023-09-08 by chipweinberger

brain9520, 2023-09-08:
> 

> **note: for iOS the initial state is typically BluetoothAdapterState.unknown**
> 
> ```
> // wait bluetooth to be on & print states
> // note: for iOS the initial state is typically BluetoothAdapterState.unknown
> // note: if you have permissions issues you will get stuck at BluetoothAdapterState.unauthorized
> await FlutterBluePlus.adapterState
>  .map((s){print(s);return s;})
>  .where((s) => s == BluetoothAdapterState.on)
>  .first;
> ```

Thanks for the tip, much appreciated.