chipweinberger/flutter_blue_plus #549: [Bug]: FlutterBluePlus.adapterState.first == BluetoothAdapterState.on is always true on Android 13

jasaw, 2023-09-08:
### FlutterBluePlus Version

1.14.24

### Flutter Version

3.13.1

### What OS?

Android

### OS Version

13

### What happened?

`FlutterBluePlus.adapterState.first == BluetoothAdapterState.on` always returns true even after turning Bluetooth off from settings. My app is now unable to detect the Bluetooth is off, so unable to warn the user.

I've just migrated from `flutter_blue` and `isOn` used to work there. `flutter_blue_plus` is not reading the on/off status correctly. I noticed `flutter_blue` uses `mBluetoothAdapter.isEnabled()` call whereas `flutter_blue_plus` uses `mBluetoothAdapter.getState()`. Does this make a difference?

### Logs

```shell
/[FBP-Android](29336): [FBP-Android] onMethodCall: flutterHotRestart
2
I/BluetoothAdapter(29336): STATE_ON
D/BluetoothLeScanner(29336): could not find callback wrapper
D/[FBP-Android](29336): [FBP-Android] disconnectAllDevices
D/[FBP-Android](29336): [FBP-Android] connectedPeripherals: 0
D/[FBP-Android](29336): [FBP-Android] HotRestart: complete
D/[FBP-Android](29336): [FBP-Android] onMethodCall: isAvailable
D/[FBP-Android](29336): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](29336): [FBP-Android] onMethodCall: getAdapterState
I/flutter (29336): 0:00:02.617666	W	bluetooth:	 *************** bluetoothIsOn: true
```

jasaw, 2023-09-08:
My quick test shows that `mBluetoothAdapter.isEnabled()` also returns true even when bluetooth is turned off. Possibly broken when I upgraded to Android targetSDK 33 from 31.

chipweinberger, 2023-09-08:
thanks for filing. if you can debug it further that would be very useful

jasaw, 2023-09-08:
Any idea what else to try? I'm debugging at the moment.

chipweinberger, 2023-09-08:
what is returned by `mBluetoothAdapter.getState();`

jasaw, 2023-09-08:
`getState()` returns value 12, which is `STATE_ON`.
I've called the `isEnabled()` in flutter_blue_plus, but it's returning true.

chipweinberger, 2023-09-08:
if `getState()` is returning `STATE_ON`, and `isEnabled()` is returning true, seems like the issue may not be in FlutterBluePlus, as you suggest.

jasaw, 2023-09-08:
Code detects the off state correctly when phone is in flight mode:
>I/BluetoothAdapter( 4516): STATE_BLE_ON
I/BluetoothAdapter( 4516): appoCount is = 0 G/S = 1/1
W/BluetoothAdapter( 4516): getBluetoothLeScanner() ble not available
D/[FBP-Android]( 4516): [FBP-Android] disconnectAllDevices
D/[FBP-Android]( 4516): [FBP-Android] connectedPeripherals: 0
D/[FBP-Android]( 4516): [FBP-Android] HotRestart: complete
D/[FBP-Android]( 4516): [FBP-Android] onMethodCall: isAvailable
D/[FBP-Android]( 4516): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android]( 4516): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 4516): [FBP-Android] getState: 10

Taking device out of flight mode and the code works properly now... Just need to bounce the bluetooth stack ???
I'm closing this as it's an Android problem.

closed 2023-09-08 by jasaw