chipweinberger/flutter_blue_plus #550: [Bug]: both iOS and Android can't reconnect after a disconnect

trabulium, 2023-09-08:
### FlutterBluePlus Version

1.14.24 (and earlier)

### Flutter Version

3.13.1

### What OS?

iOS

### OS Version

Android 10, iOS 15.7.5

### What happened?

Thanks for taking on this project and moving it along so fast as you've been doing. I've upgraded from v1.11 to 1.14.11 to 1.14.24 over the last week. 

On iOS, I keep getting: [Error] The connection has timed out unexpectedly.

In 1.14.24 my Search, Connect / Disconnect buttons do nothing after a disconnect - I have to hot restart to get them working again. The only change I made was updating from 1.14.11 to 1.14.24. I've not yet tested Android. I will test Android and then roll back and slowly roll forward to see if I can find the issue. 

I was sending updates from the MB to Flutter every 500ms, I have reduced it to 1000ms now and it seems to have slightly improved. When the disconnect happens, Bluefruit from another phone can see the MB but can't connect until I restart it which seems to indicate it's a MB problem. That said, if I do a disconnect normally, I can't reconnect in either iOS or Android until I restart the MB. Here in this URL: https://medium.com/android-news/lessons-for-first-time-android-bluetooth-le-developers-i-learned-the-hard-way-fee07646624 he speaks of using only  close(), not disconnect.

> Calling disconnect right before close: calling bluetoothDevice.disconnect() right before bluetoothDevice.close() is redundant and can result in issues in some devices. Try using only close() instead.

On top of this, I can't get your current example app to even search and find any device at all in iOS / Android.

### Logs

```shell
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse
[FBP-iOS] didUpdateValueForCharacteristic 38E12456-54EE-F6F1-A62E-1D377FA6169A
[FBP-iOS] didUpdateValueForCharacteristic 38E12456-54EE-F6F1-A62E-1D377FA6169A
flutter: Button Click: [0, 0, 64, 2, 255]
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse
[FBP-iOS] didUpdateValueForCharacteristic 38E12456-54EE-F6F1-A62E-1D377FA6169A
.....
flutter: Button Click: [0, 0, 64, 2, 255]
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse
[FBP-iOS] didUpdateValueForCharacteristic 38E12456-54EE-F6F1-A62E-1D377FA6169A
[FBP-iOS] didDisconnectPeripheral: [Error] The connection has timed out unexpectedly.
flutter: Button Click: [0, 0, 79, 2, 255]
```

trabulium, 2023-09-08:
Here's the output of trying to disconnect and then reconnect in Android. Once the second connect / disconnect happens, it's no longer visible in Bluefruit on the same device or different device.

D/[FBP-Android](16654): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](16654): [FBP-Android] onCharacteristicWrite: uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e status: 0
4
D/[FBP-Android](16654): [FBP-Android] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
D/[FBP-Android](16654): [FBP-Android] onMethodCall: disconnect
D/BluetoothGatt(16654): cancelOpen() - device: F7:E1:89:A9:CB:F9
D/BluetoothGatt(16654): onClientConnectionState() - status=0 clientIf=11 device=F7:E1:89:A9:CB:F9
D/[FBP-Android](16654): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: disconnected
D/BluetoothGatt(16654): close()
D/BluetoothGatt(16654): unregisterApp() - mClientIf=11
D/ViewRootImpl[Toast](16654): hardware acceleration = true , fakeHwAccelerated = false, sRendererDisabled = false, forceHwAccelerated = false, sSystemRendererDisabled = false
E/GraphicExt(16654): GraphicExtModuleLoader::CreateGraphicExtInstance false
D/Surface (16654): Surface::connect(this=0x79d9a20000,api=1)
D/mali_winsys(16654): EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
D/Surface (16654): Surface::setBufferCount(this=0x79d9a20000,bufferCount=3)
D/Surface (16654): Surface::allocateBuffers(this=0x79d9a20000)
D/Surface (16654): Surface::disconnect(this=0x79d9a20000,api=1)
D/View    (16654): [Warning] assignParent to null: this = android.widget.FrameLayout{e335346 V.E...... ......ID 0,0-619,115}
D/[FBP-Android](16654): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](16654): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter(16654): isLeEnabled(): ON
D/BluetoothLeScanner(16654): could not find callback wrapper
I/flutter (16654): listening
D/[FBP-Android](16654): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(16654): isLeEnabled(): ON
D/BluetoothLeScanner(16654): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
D/[FBP-Android](16654): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(16654): connect() - device: F7:E1:89:A9:CB:F9, auto: false
D/BluetoothGatt(16654): registerApp()
D/BluetoothGatt(16654): registerApp() - UUID=3c874360-f255-4fed-bfed-a46ae2e9ee63
D/BluetoothGatt(16654): onClientRegistered() - status=0 clientIf=11
D/BluetoothGatt(16654): onClientConnectionState() - status=0 clientIf=11 device=F7:E1:89:A9:CB:F9
D/[FBP-Android](16654): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
D/[FBP-Android](16654): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(16654): discoverServices() - device: F7:E1:89:A9:CB:F9
D/BluetoothGatt(16654): onConnectionUpdated() - Device=F7:E1:89:A9:CB:F9 interval=6 latency=0 timeout=500 status=0
D/[FBP-Android](16654): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/ViewRootImpl[Toast](16654): hardware acceleration = true , fakeHwAccelerated = false, sRendererDisabled = false, forceHwAccelerated = false, sSystemRendererDisabled = false
E/GraphicExt(16654): GraphicExtModuleLoader::CreateGraphicExtInstance false
D/Surface (16654): Surface::connect(this=0x79d9a20000,api=1)
D/mali_winsys(16654): EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
D/Surface (16654): Surface::setBufferCount(this=0x79d9a20000,bufferCount=3)
D/Surface (16654): Surface::allocateBuffers(this=0x79d9a20000)
D/BluetoothGatt(16654): onClientConnectionState() - status=8 clientIf=11 device=F7:E1:89:A9:CB:F9
D/[FBP-Android](16654): [FBP-Android] onConnectionStateChange: status: 8 (CONNECTION_TIMEOUT) newState: disconnected
D/BluetoothGatt(16654): close()
D/BluetoothGatt(16654): unregisterApp() - mClientIf=11
D/[FBP-Android](16654): [FBP-Android] onMethodCall: getConnectedSystemDevices

chipweinberger, 2023-09-08:
I am not reproducing this issue.

and in your logs it looks like you reconnect without issues

```
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/16654): [FBP-Android] onMethodCall: connect
...
D/[FBP-Android](https://github.com/boskokg/flutter_blue_plus/issues/16654): [FBP-Android] onConnectionStateChange: status: 0 (SUCCESS) newState: connected
```

chipweinberger, 2023-09-08:
This is most likely a problem with your bluetooth device. 

Usually when you get `The connection has timed out unexpectedly`, your device is misbehaving.

closed 2023-09-08 by chipweinberger

trabulium, 2023-09-08:
Thanks for the response.

It connects and then disconnects ~5 seconds later. Bluefruit can connect and disconnect perfectly fine from both devices, after I disconnect, I can connect my app fine but if I disconnect from the iPhone or Android, then Bluefruit can't connect anymore. 

 Additionally, in your 1.14.24, there's an issue here:

  // Disconnect from a device
  Future<void> disconnectFromDevice(_device) async {
    await _device.disconnect();
    print('Trying to disconnect: ' + _device.name);
    
When there's a disconnect, the await never returns and so print statement doesn't run. I'm about to try your latest 1.5.x that you just pushed. Can you confirm the example app is working?