chipweinberger/flutter_blue_plus #551: [Question]: When call device.createBond() application crashes

alisherbu, 2023-09-08:
### FlutterBluePlus Version

1.14.23

### Flutter Version

3.13.1

### What OS?

Android

### OS Version

Samsung Android 12

### Ask your question

But If I call device.connect(), everything work fine. To connect to device, first we should bond/pair. Is it right? However, device.connect() method is doing bond/pair and connect operations.

### Logs

```shell
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(createBond, device is disconnected, null, null)
```

chipweinberger, 2023-09-08:
You cannot create a bond with a disconnected device. 

**Question for you:** is the error confusing?

i.e. `PlatformException(createBond, device is disconnected`.

How could I make the flutter blue plus api better?

Doe `PlatformException` make it seem like FBP is misbehaving?

Would It be better if I throw this error from the dart side? as a FlutterBluePlusException?

closed 2023-09-08 by chipweinberger

alisherbu, 2023-09-08:
Wow, thank you!