chipweinberger/flutter_blue_plus #552: The "simplified" device scan API is harder to use

ekuleshov, 2023-09-08:
### FlutterBluePlus Version

1.15.0

### Flutter Version

3.13

### What OS?

All

### OS Version

n/a

### What happened?

In the 1.15.0 a "simplified" device scan API is being forced to the `FlutterBluePlus` consumers. This change also resonates with the [absence of overall API governance](https://github.com/boskokg/flutter_blue_plus/issues/447) when the major API changes being introduced out of the blue (pun intended) and are breaking all the existing plugin consumers.

Instead of a locally managed and a reasonably short-lived results stream we now have a global static scan results stream.

That makes all plugin consumers to deal with the global state and maintain their own lifecycle in the code that need to work with the scan results. Especially when there are independent parts of the application working with these scan results and they need to avoid overlapping of these results. Things are even worse when scans have to use different filters - so the global results stream is now contains the mixed results without an easy way to tell what scan request(s) these results are associated with.

I don't see any other way than register/unregister stream listeners, which significantly complicates things, comparing to the previously available scan results stream returned from now removed `FlutterBluePlus.scan()` method. That stream was managed and closed by the `FlutterBluePlus`, so no additional handling was required in the app.

### Logs

```shell
N/A
```

chipweinberger, 2023-09-08:
this should work:

```
extension Scan on FlutterBluePlus {
  static Stream<ScanResult> scan({
    List<Guid> withServices = const [],
    Duration? timeout,
    bool androidUsesFineLocation = false,
  }) {
    final controller = StreamController<ScanResult>();

    var subscription = FlutterBluePlus.scanResults.listen(
      (r) => controller.add(r.first),
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );

    FlutterBluePlus._isScanning.add(true);

    FlutterBluePlus.startScan(
      withServices: withServices,
      timeout: timeout,
      removeIfGone: null,
      oneByOne: true,
      androidUsesFineLocation: androidUsesFineLocation,
    );

    Future scanComplete = FlutterBluePlus.isScanning.where((e) => e == false).first;

    scanComplete.whenComplete(() {
      subscription.cancel();
      controller.close();
    });

    return controller.stream;
  }
}
```

chipweinberger, 2023-09-08:
> That makes all plugin consumers to deal with the global state and maintain their own lifecycle in the code that need to work with the scan results. Especially when there are independent parts of the [...]

There can only be a single scan in progress at one time. This is a limitation of Android & iOS.

ekuleshov, 2023-09-08:
> This is how all the other streams work, too.`onValueReceived `, `mtu`, `connectionState`, `adapterState`, `isScanning`, `bondState`, etc. The stream controllers are global, and expose a single forever-lived stream.

A global stream for adapter state does make sense. But otherwise global state is bad.

Though I was under impression that value received, mtu, connection and bond state - are device/connection specific (i.e. not global). Otherwise you can't have multiple connected devices at the same time, or am I missing something?

It would be more flexible and simpler to work with if each scan would be a separate entity and had its own results and a lifecycle. Kind of similar to a connected device.

> Scans cannot use different filters. 

Eh? The `scan()` API does allow to specify different latency or services...

> There can only be a single scan in progress at one time. Mixing filters is not possible. This is a limitation of Android & iOS.

I know there can be only one scan at a time. But you can spawn scans with different filters. But I don't see why each individual scan can't have its own listeners or how it is a better solution to cancel an ongoing scan, e.g. comparing, for example, to queuing scans to run one after another (in most cases each scan has sort time to live).

The old API was giving options to consumers of the plugin to use global results or subscribe to a given scan. So they won't have to use boiler plate utility methods like you provided. Also your method has some gaps and timing issues that may not easy to address properly, e.g. the listener you registering on the scanResults could get events even before a new scan is started.

Also, you are making consumers to rely on the private state of plugin and non-API fields using code like `FlutterBluePlus._isScanning.add(true);` - generally that is fragile, not an advisable solution and it may not pass on the strict code liner checks.

Similarly. Now it is unclear what happens with an ongoing scan if someone else triggers a new scan. The API change says that the 1st scan will be canceled when a 2nd scan is started - but how does the application can get notified on the 1st scan being canceled or sort out which results are which?

PS: I also don't quite get notes about removing some properties or features that only supported on Android or on iOS. There are still parameters like `androidUsesFineLocation`, so having some feature not supported on one platform is not necessarily a reason to not have it on all platforms.

chipweinberger, 2023-09-08:
> or am I missing something?

the streams all use the same global broadcast StreamController. But yes, they are 'device' specific in terms of api.

>  Also your method has some gaps and timing issues

thanks! I just fixed that. Like the old api, it now throws an exception if already scanning. Yes, using `FlutterBluePlus._isScanning.add(true)` is not ideal but I have no plans to remove that ever.

>  how does the application can get notified on the 1st scan being canceled or sort out which results are which?

you can use `FlutterBluePlus.isScanning` to be notified when a scan ends. You seem very smart, so I'm sure you can figure out a way to determine which are which.

---

Overall, yes I understand you do not want breaking changes, and I get that. But as the maintainer, I spend most of my time helping people that are confused by the API, so it is a bit unfair for me. Simplifying the API is important.

Overall, despite ~500 commits I've added to this project, there have been very few breaking changes and all have been rather easy for consumers to address. So maybe you should say thank you a little bit too!

chipweinberger, 2023-09-09:
@ekuleshov , have you tried the scan extension in your code yet?

ekuleshov, 2023-09-09:
> @ekuleshov , have you tried the scan extension in your code yet?
> It should be a drop in replacement for your existing calls to `scan`.

I still find it gross to require such extension in every app where I need BLE functionality, given that the previously available API was providing functionality out of the box. I'll have to look at the required application changes again when I have some free time.

Also not sure if this extension handles all cases. E.g. what happens when I call `close()` on the returned `Stream`?
Generally, the `Steam` API is an out of the box Dart abstraction and BLE scan functionality could have been designed around it, vs stitching together multiple static methods, a single global state and potentially having some gaps or corner cases in the implied state machine between these static methods. Some I already mentioned, such as the scan results from different scans being mixed together.

As an alternative, a `scan()` call that spawns a separate Stream would be a self contained API based on the common Dart's async concepts and a well know lifecycle (started, completed, canceled, etc) and would allow to handle corner cases more naturally - e.g. starting a new scan could just complete the previously created scan stream, so its listeners could handle that completion, etc. And that way each scan steam will also only have its own results.

> Overall, yes I understand you do not want breaking changes, and I get that. But as the maintainer, I spend most of my time helping people that are confused by the API, so it is a bit unfair for me. Simplifying the API is important.

It is not that I don't want breaking changes... I would like to see changes that are well justified or at least had some kind of discussion or governance up front. But right now it looks like a lot of the changes are done based on your personal preferences or views or from the point of view of the plugin maintenance.

Though simplicity of the maintenance should not be the goal. You need to consider the convenience of use and the needs of the applications to drive the development. Hence why I suggested to create some kind of governance process or steering committee for the plugin that would bring some order into the ongoing plugin development. Unfortunately that request was dismissed as unimportant.

I do understand you putting a lot of time into this project. However you really need to look back at these 500 commits. See how many regressions have been introduced in these commits. Basically the plugin is unsafe to update without our own rigorous testing, and generally we don't know what else was broken in the ongoing changes or what version is safe to use.

chipweinberger, 2023-09-10:
i'm toying with the idea of adding it back.

but im not sure how many people actually use this api since the readme encourages startScan.

but I would want to name it better. `scan` vs `startScan` is confusing imo.

chipweinberger, 2023-09-10:
Another idea, 

perhaps we can have scan as an internal method i.e. _scan

if users want to use it they can define an extension to access it 

overall, the idea of having public 2 api to start a scan is kind of ridiculous imo. neither apple nor google have multiple api for starting a scan in their code.

FritzMatthaeus, 2023-09-11:
Hi Chip,

I am not sure, if this is perfectly related to this issue. 

I am not sure it is a good solution to handle FlutterBluePlus without instantiating on static method and properties only. As far as i understood the conversation, this change to static-only methods does break a lot of implementations. My argument is slightly different. 

For better testing and development, i want to mock the FlutterBluePlus instance, e.g. using **Fake**. Ideally there would be an interface to make sure, that alle FlutterBluePlus methods/properties are properly overriden. Even without interface we can mock it with Fake and get an IDE-Support (i am using VSCODE) for overriding methods/properties. This is helpful to get errors in the IDE on changes in the FlutterBluePlus package and adjust tests. 

If working with static methods/properties only, there is no IDE Support for overriding methods/properties. On every package change, i will have to manually investigate if anything changed. There are workaround solutions: 
- I could mock the method channels used by the package, but this implies a lot of state management logic
- I can write a Wrapping Class for FlutterBluePlus and map all static to instance properties/methods. But this extra work is not necessary if working with a FlutterBluePlus instance.

That's why i would prefer to stick to an instance (as it has been previously) and make FlutterBluePlus a Singleton to prevent multiple instantiation. 

What do you think? 
In case the package will stick with static, any suggestions on setting up the Mock?

Thanks for your work and best regards,
Fritz

ekuleshov, 2023-09-11:
The `instance` field issue is not really related to these `scan()` changes, but I do agree, having everything static is not a flexible approach and prevent stabbing or mocking of the BLE services. Multiple static methods stitched together are even harder to stub, because you have to emulate implementation details, which is generally fragile and a bad idea.

chipweinberger, 2023-09-16:
for people reading this in the future, see [MIGRATION.md](https://github.com/boskokg/flutter_blue_plus/blob/master/MIGRATION.md)

closed 2023-09-16 by chipweinberger

chipweinberger, 2023-09-25:
@ekuleshov did the extension work for you? Or did you migrate your code?

songyan-microprogram, 2023-09-28:
@chipweinberger It is hoped that with every occurrence of breaking changes, an update to a new major version can be made. Failing to do so may result in projects, which depend on this package indirectly through other packages, being unable to rectify errors promptly.

chipweinberger, 2023-09-28:
That's fair. I'm not a fan of that system personally as it does not reflect how much actually changed. 

Instead, I use the 1.X.0 to reflect breaking changes. Downstream users should lock to a specific 1.X version, i.e. `1.15.X^`

Perhaps I'll document this in the README.

ekuleshov, 2023-11-04:
> @ekuleshov did the extension work for you? Or did you migrate your code to `startScan`?

First of all the "extension" part is misleading. You can't extend Dart classes with static methods. Basically that is some static method you have to copy-paste into your code.

In any case the proposed code is failing with a "Bad state: No element" exception here:

```
StreamSubscription<List<ScanResult>> subscription = FlutterBluePlus.scanResults.listen(
      (r) => controller.add(r.first), // <---- r.first does not exist here
      onError: (e, stackTrace) => controller.addError(e, stackTrace),
    );
```

PS: I'm now stuck with several issues/bugs in the old plugin version and the only option is look for the version where they are maybe fixed (and possibly introduced other issues).

chipweinberger, 2023-11-04:
> In any case the proposed code is failing with a "Bad state: No element" exception

i forgot that we emit an empty array as our first element 

just check that there are elements before you call first.

ekuleshov, 2023-11-04:
> just check that there are elements before you call first.

Obviously. However I still can't upgrade, as there some other hidden breaking or behavioural changes. With new version BLE devices disconnects out of the blue, BLE notifications aren't being delivered and "write" not supported errors are thrown on a characteristics that are writable (at least on macos). I will have to put it on hold again for some better time.

chipweinberger, 2023-11-05:
please open bug reports with all the relevant details 

the goal is to make this library completely bug free