chipweinberger/flutter_blue_plus #553: [Bug]: Extension conflict on firstWhereOrNull

leijdekkers, 2023-09-10:
### FlutterBluePlus Version

1.15.4

### Flutter Version

3.13.3

### What OS?

All

### OS Version

ios 16

### What happened?

I am using the extension "firstWhereOrNull" using the dart  collections library. It looks like flutter_blue_plus also defines exactly the same extension.  Maybe it would be better to remove the "firstWhereOrNull" extension from the flutter blue plus library and use the once as defined by dart in collections library?

Below the compiler error:

```
Try using an explicit extension application of the wanted extension or hiding unwanted extensions from scope.
    final bin = auditedBins.where((bin) => bin.name == binName && bin.binRoomNr == binRoomNr).firstWhereOrNull((bin) => bin.count > 0);
                                                                                          ^^^^^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dev/collection-1.18.0/lib/src/iterable_extensions.dart:244:6: Context: This is one of the extension members.
  T? firstWhereOrNull(bool Function(T element) test) {
     ^^^^^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dev/flutter_blue_plus-1.15.4/lib/src/utils.dart:441:6: Context: This is one of the extension members.
  T? firstWhereOrNull(bool Function(T) test) {
     ^^^^^^^^^^^^^^^^
```

### Logs

```shell
none
```

chipweinberger, 2023-09-10:
fixed 1.15.5

closed 2023-09-10 by chipweinberger

leijdekkers, 2023-09-10:
Perfect, thanks, compiles again!

ekuleshov, 2023-09-11:
I was about to report the same issue. The problem is that conflict was introduced in 1.14.x and this fix is being made in 1.15.x which has additional breaking changes. Sigh...