chipweinberger/flutter_blue_plus #554: [Feature]: Let startScan return List<ScanResult>

leijdekkers, 2023-09-10:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.13.3

### What OS?

All

### OS Version

ios 16

### Feature Request

I know that you removed the List<ScanResult as a future for the startScan but i find it very useful to have! In my app i am looking for a specific bluetooth device and once i have it i return thet device. Code looks like this

```
  List<ScanResult> results = await FlutterBluePlus.startScan( timeout: timeOut, withServices: [serviceId]);
  ScanResult? result = results.where((result) => result.device.remoteId.str == remoteId).firstOrNull;
```

I agree that startScan is maybe not the best name for it but for me it is very useful to have the await functionality instead of using "FlutterBluePlus.scanResults.listen((results)...." way of getting the devices. Any chance you can put it back and maybe call startScan method differently.

chipweinberger, 2023-09-10:
This should work. let me know if you have problems and i'll update it

```
extension Scan on FlutterBluePlus {
  static Future<List<ScanResult>> startScanWithResult({
    List<Guid> withServices = const [],
    Duration? timeout,
    bool androidUsesFineLocation = false,
  }) async {
    if (FlutterBluePlus.isScanningNow) {
      throw Exception("Another scan is already in progress");
    }

    List<ScanResult> output = [];

    var subscription = FlutterBluePlus.scanResults.listen((result) {
      output = result;
    }, onError: (e, stackTrace) {
      throw Exception(e);
    });

    FlutterBluePlus.startScan(
      withServices: withServices,
      timeout: timeout,
      removeIfGone: null,
      oneByOne: false,
      androidUsesFineLocation: androidUsesFineLocation,
    );

    // wait scan complete
    await FlutterBluePlus.isScanning.where((e) => e == false).first;

    subscription.cancel();

    return output;
  }
}
```

chipweinberger, 2023-09-10:
that said, I would recommend switching to `FlutterBluePlus.scanResults.listen((results)` because you can dramatically lower the time it takes to find the device you want.

leijdekkers, 2023-09-10:
Thanks for the quick fix! Your extension works fine. For my application it does not really matter that it takes a bit longer to find the device. It is still fairly quick. When i have more time i will refactor my code to use "listen" and then handle the result properly.

chipweinberger, 2023-09-10:
great!

please keep ths issue open for others to find.

chipweinberger, 2023-09-10:
when you refactor, i recommend using `oneByOne` mode, and awaiting the first result that you are looking for. 

something like this (i think)

```
BluetoothDevice? device = FlutterBluePlus.scanResults
    .map((list) => list.first)
    .where((r) => r.device.localName == "myDeviceName")
    .firstOrNull;

await FlutterBluePlus.startScan(timeout: Duration(seconds: 10), oneByOne:true);
```

leijdekkers, 2023-09-10:
Yes i can confirm that the code above works! Thanks!

dlewis2017, 2023-09-14:
That does work, thanks!! 

I'm having some trouble with re-running the scan though, can you confirm re-scanning works?

chipweinberger, 2023-09-16:
if you're reading this in the future, see [MIGRATION.md](https://github.com/boskokg/flutter_blue_plus/blob/master/MIGRATION.md)

closed 2023-09-16 by chipweinberger