chipweinberger/flutter_blue_plus #555: [Bug]: app crash, when I changed bluethooth permissions in app settings

bastengao, 2023-09-10:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.13.3

### What OS?

iOS

### OS Version

16.6

### What happened?

1. I call these two methods

```dart
FlutterBluePlus.adapterState.listen((state) async {
    if (state == BluetoothAdapterState.on) {
        try {
          await FlutterBluePlus.startScan(oneByOne: false, withServices: [Guid(serviceUUID)]);
        } catch(e) {
          print("startScan error: $e");
        }
      } else if (state == BluetoothAdapterState.off) {
        try {
          await FlutterBluePlus.stopScan();
        } catch(e) {
          print("stopScan error: $e");
        }
      }
}
FlutterBluePlus.scanResults.listen
```

2. turned bluethooth permission on in app settings

3. app crashed

### Logs

```shell
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGKILL
    frame #0: 0x00000001c9b9aca4 libsystem_kernel.dylib`mach_msg2_trap + 8
libsystem_kernel.dylib`mach_msg2_trap:
->  0x1c9b9aca4 <+8>: ret
libsystem_kernel.dylib`macx_swapon:
    0x1c9b9aca8 <+0>: mov    x16, #-0x30
    0x1c9b9acac <+4>: svc    #0x80
    0x1c9b9acb0 <+8>: ret
Target 0: (Runner) stopped.
Lost connection to device.
```

chipweinberger, 2023-09-10:
this is a low level crash

could be Dart vm issue, or OS issue.

chipweinberger, 2023-09-11:
going to close this since there is nothing we can do

closed 2023-09-16 by chipweinberger