chipweinberger/flutter_blue_plus #556: [Question]:

FritzMatthaeus, 2023-09-11:
### FlutterBluePlus Version

1.15.0

### Flutter Version

3.13

### What OS?

All

### OS Version

all

### Ask your question

Hi Chip,

I am not sure, if this is perfectly related to this issue.

I am not sure it is a good solution to handle FlutterBluePlus without instantiating on static method and properties only. As far as i understood the conversation, this change to static-only methods does break a lot of implementations. My argument is slightly different.

For better testing and development, i want to mock the FlutterBluePlus instance, e.g. using Fake. Ideally there would be an interface to make sure, that alle FlutterBluePlus methods/properties are properly overriden. Even without interface we can mock it with Fake and get an IDE-Support (i am using VSCODE) for overriding methods/properties. This is helpful to get errors in the IDE on changes in the FlutterBluePlus package and adjust tests.

If working with static methods/properties only, there is no IDE Support for overriding methods/properties. On every package change, i will have to manually investigate if anything changed. There are workaround solutions:

    I could mock the method channels used by the package, but this implies a lot of state management logic
    I can write a Wrapping Class for FlutterBluePlus and map all static to instance properties/methods. But this extra work is not necessary if working with a FlutterBluePlus instance.

That's why i would prefer to stick to an instance (as it has been previously) and make FlutterBluePlus a Singleton to prevent multiple instantiation.

What do you think?
In case the package will stick with static, any suggestions on setting up the Mock?

Thanks for your work and best regards,
Fritz

### Logs

```shell
no logs, just discussion
```

chipweinberger, 2023-09-11:
> I can write a Wrapping Class for FlutterBluePlus and map all static to instance properties/methods.

This is the approach I would recommend.

FlutterBluePlus api should be pretty stable at this point and has a pretty small api.

Would only take a few minutes of work.

FritzMatthaeus, 2023-09-11:
Then i'll have to go that way. I am not really convinced by your design choice, but hey, there myriads of ways to get to Rome. Thx for your work.

FritzMatthaeus, 2023-09-12:
Sorry to address this question again, but this static way is really not convenient for mocking and testing. 

A Wrapper class that maps static to instance properties, as you recommended, is not possible. One cannot access static methods from an instance:

```
class Wrapper {
  final FlutterBluePlus flutterBluePlus;

  Wrapper({required this.flutterBluePlus});

  Stream<List<ScanResult>> scan() {
    return flutterBluePlus.scanResults; // not possible with static methods / properties
  }
}
```

As alternative i have to write a switcher on every getter and method, which is not a nice code, in my opinion:

```
class GreentankBleAdapter {
  final bool isMock;

  GreentankBleAdapter({this.isMock = false});

  Stream<List<ScanResult>> scan() {
    return isMock ? FakeFlutterBluePlus.scanResults : FlutterBluePlus.scanResults;
  }
}
```

I think that this Wrapper does not have to know or care if the App is mocked or not.

FritzMatthaeus, 2023-09-12:
The scan() method is a good example for the downside of static methods. I mocked it in my FakeFlutterBluePlus but now it is deprecated and throwing an Exception as startScan() should be used. There is no way of knowing this in my FakeFlutterBluePlus. 

An interface implemented by a FlutterBluePlus instance would be really helpful in this case as i could implement it in my FakeFlutterBluePlus and get IDE Support on any changes.

daniloapr, 2023-09-15:
> Hi Chip,
>
> I am not sure, if this is perfectly related to this issue.
>
> I am not sure it is a good solution to handle FlutterBluePlus without instantiating on static method and properties only. As far [...]

Totally agree!!

chipweinberger, 2023-09-15:
> One cannot access static methods from an instance

your wrapper methods should be static.

and you should be wrapping the static FBP calls.

closed 2023-09-16 by chipweinberger

FritzMatthaeus, 2023-09-18:
Hi Chip,

the only way for me to mock this, is writing a FakeFlutterBluePlus mocking all the static methods of FlutterBluePlus. But then i end up with the question, how to use this FakeFlutterBluePlus. The only way i came up with is the one mentioned above: 

```
// my fake FlutterBluePlus class
class FakeFlutterBluePlus {

  static final CustomStreamController<List<FakeScanResult>> _scanResultsList =
      CustomStreamController(initialValue: []);

  static Stream<List<FakeScanResult>> get scanResults =>
      _scanResultsList.stream;
}

class Wrapper {
  final bool isMock;

  Wrapper({this.isMock = false});

   /// returns a stream of [ScanResult]s
  Stream<List<ScanResult>> get scanResults =>
      isMock ? FakeFlutterBluePlus.scanResults : FlutterBluePlus.scanResults;
}

/// Class handling all Bluetooth related use cases
class BluetoothRepo{
  final Wrapper wrapper;

 BluetoothRepo(this.wrapper);

   /// returns a stream of [ScanResult]s
  Stream<List<ScanResult>> get scanResults =>
      wrapper.scanResults;
}
```

I have a Repo class connecting my usecases with the bluetooth device. To mock it, i cannot use FlutterBluePlus directly, as i would have to rewrite all FlutterBluePlus.calls when mocking it. So i wrote an additional wrapper class that does this via a flag. When mocking i just initialze the Wrapper class with a different flag. 

It works, but it's not really nice. Would be so much nicer to do this:

```
class FakeFlutterBluePlus extends Fake implements FlutterBluePlus {

  @override
  Stream<List<FakeScanResult>> get scanResults =>
      [];
}

/// Class handling all Bluetooth related use cases
class BluetoothRepo{
  final FlutterBluePlus fbp; // can be FlutterBluePlus or FlutterBluePlusFake

 BluetoothRepo(this.fbp);

   /// returns a stream of [ScanResult]s
  Stream<List<ScanResult>> get scanResults =>
      fbp.scanResults;
}

```

This way has 2 advantages: 
1) i do not need an extra wrapper class
2) i get IDE Support on my FakeFlutterBluePlus class if FlutterBluePlus has changes in properties or methods

chipweinberger, 2023-10-16:
how to mock: https://dsavir-h.medium.com/mocking-bluetooth-in-flutter-updated-cb3b9484ae02

see: https://github.com/boskokg/flutter_blue_plus/blob/master/MOCKING.md