chipweinberger/flutter_blue_plus #557: [Bug]: Logs spam the console even when logLevel is set to None

nskAdhawk, 2023-09-11:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

13

### What happened?

We get a stream of logs from our peripheral, so `onCharacteristicsChanged` gets printed continuously in the console, obscuring everything else. 

`FlutterBluePlus.setLogLevel(LogLevel.none)` is the first thing called. 

This wasn't an issue on 1.14.18.

### Logs

```shell
D/[FBP-Android](26611): [FBP-Android] onMethodCall: flutterHotRestart
D/BluetoothAdapter(26611): isLeEnabled(): ON
D/[FBP-Android](26611): [FBP-Android] disconnectAllDevices
D/[FBP-Android](26611): [FBP-Android] connectedPeripherals: 0
D/[FBP-Android](26611): [FBP-Android] HotRestart: complete
D/[FBP-Android](26611): [FBP-Android] onMethodCall: setLogLevel
D/[FBP-Android](26611): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter(26611): isLeEnabled(): ON
D/BluetoothLeScanner(26611): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
D/[FBP-Android](26611): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](26611): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](26611): [FBP-Android] onMethodCall: getConnectedSystemDevices
D/[FBP-Android](26611): [FBP-Android] onMethodCall: connect
D/BluetoothGatt(26611): connect() - device: EC:4D:E5:34:43:0D, auto: true
D/BluetoothGatt(26611): registerApp()
D/BluetoothGatt(26611): registerApp() - UUID=733d9dca-d294-476c-b01d-33a58c05ce3b
D/BluetoothGatt(26611): onClientRegistered() - status=0 clientIf=7
D/[FBP-Android](26611): [FBP-Android] onMethodCall: getConnectedSystemDevices
```
```

chipweinberger, 2023-09-15:
you should open a PR if you can fix it

chipweinberger, 2023-09-16:
I dont see any problems in the code. we only `onCharacteristicChanged` in debug

```
        @Override
        public void onCharacteristicChanged(BluetoothGatt gatt, BluetoothGattCharacteristic characteristic)
        {
            // this callback is only for notifications & indications
            log(LogLevel.DEBUG, "[FBP-Android] onCharacteristicChanged: uuid: " + uuid128(characteristic.getUuid()));
```

closed 2023-09-16 by chipweinberger

nskAdhawk, 2023-10-05:
Sorry for the delay. I looked into it a little further. 
I didn't copy the `onCharacteristicsChanged` log messages, but you can see that there are other debug level messages being printed.

The first method that I call is `FlutterBluePlus.setLogLevel(LogLevel.none);`
It looks like the logLevel is getting overridden in `_initFlutterBluePlus`

```
I/FA      (28363): Application backgrounded at: timestamp_millis: 1696521771288
I/flutter (28363): ----------- flutter_blue_plus.dart setLogLevel LogLevel.none
I/flutter (28363): ----------- flutter_blue_plus.dart initFlutterBluePlus 
I/flutter (28363): ----------- flutter_blue_plus.dart setLogLevel LogLevel.debug
D/BluetoothAdapter(28363): isLeEnabled(): ON
D/BluetoothLeScanner(28363): could not find callback wrapper
D/[FBP-Android](28363): [FBP-Android] disconnectAllDevices
D/[FBP-Android](28363): ============= setLogLevel to 0
D/[FBP-Android](28363): ============= setLogLevel to 4
```

chipweinberger, 2023-10-05:
thanks for looking into it. Can you open a PR? thanks

chipweinberger, 2023-10-05:
fixed in 1.16.4