chipweinberger/flutter_blue_plus #558: [Bug]:

umuttsaltan, 2023-09-12:
### FlutterBluePlus Version

1.14.24

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 11

### What happened?

I'm using notify functions from different characteristics in my app. After the latest updates, it seems that I can only start the notify process on one characteristic at a time when I try to connect to multiple characteristics in same time. I can't receive data all from the characteristics. I tried to get logs but no logs there. This problem didn't exist in the 1.14.23 update but i cant downgrade because 1.14.24 solve my another problem in same time.
**Notify Function **
`void currentChar() async {
    final currentServiceUuid = Guid("0000181a-0000-1000-8000-00805f9b34fb");
    final currentCharacteristicUuid =
        Guid("96ccb2fe-ab78-11ed-afa1-0242ac120002");
     
    for (var service in variables.servicesList) {
      if (service.uuid == currentServiceUuid) {
        BluetoothCharacteristic? currentCharacteristic;

        for (var characteristic in service.characteristics) {
          if (characteristic.uuid == currentCharacteristicUuid) {
            currentCharacteristic = characteristic;
            break;
          }
        }

        try {
          await currentCharacteristic
              ?.setNotifyValue(currentCharacteristic.isNotifying == false);
        } catch (e) {
          print("current char $e");
        }

        currentCharacteristic?.lastValueStream.listen((current) {
          if (current.isEmpty) {
            print('Current  Empty');
          } else {
            if (current.length == 2) {
              int currentdata = (current[0] << 8) + current[1];
              print('current: $currentdata');
              dataDeviceController.currentData[variables.currentDeviceIndex!] =
                  currentdata.toString();
            }
          }
        });
      }
    }
  }`

### Logs

```shell
I/flutter (12270): [FBP] <setLogLevel> args: 4
I/flutter (12270): [FBP] <setLogLevel> result: null
```

umuttsaltan, 2023-09-12:
And i tried the example app, there is no problem. It looks i can connect multiple characteristics and get data. But i dont get it, what am i doing wrong ?

umuttsaltan, 2023-09-12:
I tried my app on iOS now. There is no problem. Why android have these problems? Not working properly.

vicasystem, 2023-09-13:
I think that issue 561 apply also to yours: https://github.com/boskokg/flutter_blue_plus/issues/561
Could you verify?

umuttsaltan, 2023-09-13:
> I think that issue 561 apply also to yours: #561 Could you verify?

Mate, it didnt work for me. I tried your solution for write function but still have problem. Also have notify problems on Android. I cant get data from characteristics.

chipweinberger, 2023-09-16:
if the example app works, then the problem must be in your own code. 

btw, you should setup `listen` *before* you call `setNotifyValue` to make sure you don't miss any data

closed 2023-09-16 by chipweinberger