chipweinberger/flutter_blue_plus #559: [Bug]:

changewhite, 2023-09-12:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 13

### What happened?

Characteristics are sending duplicate notifications when device is reconnected
[image]

### Logs

```shell
No error log output
```

chipweinberger, 2023-09-16:
logs are needed. Ideally verbose logs. 

please reopen when you have logs.

I'm not aware of any code in FBP that would cause this.

closed 2023-09-16 by chipweinberger

chipweinberger, 2023-09-16:
you need to stop listening to the stream (i.e. call `cancel`) when you disconnect

```
  final subscription = stream.listen((data) {
    print('Data: $data');
  });

  await subscription.cancel();
```