chipweinberger/flutter_blue_plus #561: [Bug]: Starting from 1.14.20 reading and writing characteristics doesn't work on Android (while is it ok on Ios)

vicasystem, 2023-09-13:
### FlutterBluePlus Version

1.14.20 thru 1.15.5

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

Android 11 and 12

### What happened?

We start listen on notify from characteristic with short uuid 6000. The notify value is used to obtain the characteristic to read with the updated information. Starting from 1.4.20 the characteristic is read sometimes only the first time then read seems stuck in the mutex take. No informations from log. Also, the write to some characteristic also fails for the same reason.

Please note that everything works fine on 1.14.19. Problems on **Android** starts with 1.4.20.
**On Ios everything works fine even in 1.15.5.**

Here is the code:
```
/* Function for listen on notify on BluetoothCharacteristic with uuid 6000 (passed as parameters from discoverServices)*/

Future<StreamSubscription<List<int>>> _bleStartNotifySubscription(String uuid, String mac, BluetoothCharacteristic characteristic) async {           
    await characteristic.setNotifyValue(true);
    return characteristic.lastValueStream.listen((lastValue) async{ 
      if (lastValue.isEmpty) {
        return;
      }                    
      String data = const Utf8Decoder().convert(lastValue);
      debugPrint('BLE Listen $uuid: $data');
      String payload = '';
      String uuidRead = '';
      switch(data.toLowerCase()) {
        case 'sauna': //5000 stato sauna
          uuidRead = '5000';
          break;
        case 'bturco': //5001 stato bagno turco
          uuidRead = '5001';
          break;
        case 'stato': //5002 stato generale
          uuidRead = '5002';
          break;
        default:
          return;
      }
      payload = await _bleRepository.read(mac, uuidRead);

      if (payload.isEmpty) return;
      debugPrint('BLE Read uuid $uuidRead: $payload');
      _bleRepository.setReceivedPayload(payload);
    });               
  }

/* Function in _bleRepository to read the characteristic */
  Future<String> read(String mac, String uuid) async {

    BluetoothDevice? bleDevice = getBleDevice(mac); //get stored BluetoothDevice associated at the MAC address (serial number, not Bluetooth mac address)
    if (bleDevice == null) {
      return '';
    }
    for (BluetoothService service in bleDevice.servicesList ?? []) {
      if (service.uuid == Guid('00004462-0000-1000-8000-00805f9b34fb')) {
        for (BluetoothCharacteristic characteristic in service.characteristics) {
          if (characteristic.uuid ==  Guid('0000$uuid-0000-1000-8000-00805f9b34fb')) {           
            List<int> data = await characteristic.read();
            if (data.isEmpty) {
              return '';
            }
            String msg = const Utf8Decoder().convert(data);
            //debugPrint('BLE read uuid: $uuid from $mac: $msg');
            return msg;
          }
        }
      }
    }
    return '';
  }

/* Function to publish (write) to a characteristic */
  void publish(String mac, String topic, String payload) async {
    BluetoothDevice? bleDevice = getBleDevice(mac);
    if (bleDevice == null) {
      return;
    }
    for (BluetoothService service in bleDevice.servicesList ?? []) {
      if (service.uuid == Guid('00004462-0000-1000-8000-00805f9b34fb')) {
        for (BluetoothCharacteristic characteristic in service.characteristics) {
          if (characteristic.uuid ==  Guid('00006000-0000-1000-8000-00805f9b34fb')) {
            String data = topic;
            data += payload;            
            characteristic.write(const Utf8Encoder().convert(data));
            debugPrint('BLE send command: $data');
          }
        }
      }
    }
  }
```

### Logs

```shell
We have try to set logLevel both to verbose and debug but nothing is printed. Read and Write silently fails. With stepping debug we have found that the function remains stuck on mutex take.
```

vicasystem, 2023-09-13:
After some further test I can confirm the problem is on:
 `await readMutex.take();`
in bluetooth_characteristic.dart : 75 that, after the first time, blocks forever.

My work around was to modify the mutex key in order to allow a single read per-device (and not also per-characteristic):
```
  /// read a characteristic
  Future<List<int>> read({int timeout = 15}) async {
    // check connected
    if (FlutterBluePlus._isDeviceConnected(remoteId) == false) {
      throw FlutterBluePlusException(ErrorPlatform.dart, "readCharacteristic",
        FbpErrorCode.deviceIsDisconnected.index, "device is not connected");
    }

    // Only allows a single read to be underway at any time, per-characteristic, per-device.
    // Otherwise, there would be multiple in-flight reads and we wouldn't know which response is which.
    String key = remoteId.str + /*":" + characteristicUuid.toString() +*/ ":readChr";
    /*                                         ^^^^^^^^^^^^^^^^^^^^^^ */
    _Mutex readMutex = await _MutexFactory.getMutexForKey(key);
    await readMutex.take();

```
After apply the workaround also write characteristic start to work.
Sadly, I have not found a reason why in 1.14.19 it works, as the code is the same.
In IOS probably is the OS itself that limit read and write request per-device.

I think the same solution apply to issue 558: https://github.com/boskokg/flutter_blue_plus/issues/558

vicasystem, 2023-09-13:
Sadly the workaround is not consistent. When the app start the bluetooth connection sometimes still doesn't work as in the filed bug.

chipweinberger, 2023-09-15:
let me know if you figure it out.

full logs would be useful.

chipweinberger, 2023-09-17:
closing until you have full logs.

closed 2023-09-17 by chipweinberger

qawitherev, 2023-09-18:
this bug also happened to me 
sample code: 
`void provisionTest() async {`
    `await getServiceList();`
    `getCharacteristics();`
    `final ourCharacteristics = characteristicsList[0];`
    `final canRead = ourCharacteristics.properties.read;`
    `print("QAWI3 - can read: $canRead");`
  `}`

on android (12), 
[image]

on ios (ios 16.5)
[image]

i was able to read/write into characteristics on ios, but not read/write the same characteristics on Android