chipweinberger/flutter_blue_plus #562: [Question]: discoverServices Time out after 15s

mpillawa, 2023-09-13:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.13.0

### What OS?

Android

### OS Version

13

### Ask your question

After connecting a device I try to discover the services, but I do get a Time out. 
The connect seems to work properly, the log shows "connected".

Any hints?

`        
await selectedDevice!.connect();
List<BluetoothService> services = await selectedDevice!.discoverServices();
`

Thanks.

### Logs

```shell
FlutterBluePlusException: discoverServices: (code: 1) Timed out after 15s
```

shuizhuyu1015, 2023-09-13:
same issue！
It appears after turning off the Bluetooth adapter, and then turning on the adapter to rescan.
The log shows “already connected”，but no device is actually connected.

mpillawa, 2023-09-15:
BTW: same code works on iOS

chipweinberger, 2023-09-15:
post full logs

chipweinberger, 2023-09-17:
should be fixed in 1.15.7

open a new issue with full logs if not.

closed 2023-09-17 by chipweinberger