chipweinberger/flutter_blue_plus #563: [Bug]: MacAddresses filtering when scanning

dlewis2017, 2023-09-13:
### FlutterBluePlus Version

1.15.2

### Flutter Version

3.13.0

### What OS?

iOS

### OS Version

iPadOS 16.6.1

### What happened?

More of a feature request than bug but - The scan function that allows filtering by MacAddresses is deprecated. Could the mac address filtering be added back in for developer testing when multiple Bluetooth devices with the same build are being tested by different developers? Thanks.

### Logs

```shell
https://github.com/boskokg/flutter_blue_plus/blob/49066942fbacd324c46b58b9f062ee125535f703/lib/src/flutter_blue_plus.dart#L371
```

chipweinberger, 2023-09-15:
it is easy to filter by mac address in your own code, so I do not plan to add this feature.

but thank you for filing :)

chipweinberger, 2023-09-16:
if you're reading this in the future, see [MIGRATION.md](https://github.com/boskokg/flutter_blue_plus/blob/master/MIGRATION.md)

closed 2023-09-16 by chipweinberger

dlewis2017, 2023-09-19:
@chipweinberger I wanted to circle back to this because it seems the remoteId associated with the device is a 128-bit uuid. Mac Addresses seem to have a [different format](https://www.cmu.edu/computing/services/endpoint/network-access/mac-address.html) Is remoteId definitely where the macAddress exsits?

chipweinberger, 2023-09-19:
yes use remoteId

macAddress was just a bad name imo.

chipweinberger, 2023-11-09:
this has been added back, btw. use 1.26.0+