chipweinberger/flutter_blue_plus #564: [Bug]: "disconnect" does not take effect after the adapter is turned off

shuizhuyu1015, 2023-09-14:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.3.2

### What OS?

iOS

### OS Version

iOS 16.6

### What happened?

When the device is connected, the Bluetooth adapter of the mobile phone is turned off, and the device is disconnected. 
However, when the Bluetooth adapter is turned on again and rescan, the app shows that it is connected, but the device is not actually connected, because the connected Bluetooth icon is not displayed on the device.

### Logs

```shell
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOff
[FBP-iOS] disconnected from device DEA8742F-E5B1-3879-EBB6-A07A7C92B485
[FBP-iOS] disconnectAllDevices
[FBP-iOS] calling disconnect: DEA8742F-E5B1-3879-EBB6-A07A7C92B485
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x280fa7a20> can only accept this command while in the powered on state
[FBP-iOS] handleMethodCall: stopScan
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x280fa7a20> can only accept this command while in the powered on state
[FBP-iOS] centralManagerDidUpdateState CBManagerStatePoweredOn
[FBP-iOS] handleMethodCall: getConnectedSystemDevices
[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] handleMethodCall: getConnectedSystemDevices
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: getAdapterState
[FBP-iOS] handleMethodCall: connect
[FBP-iOS] already connected
[FBP-iOS] handleMethodCall: readRssi
[CoreBluetooth] API MISUSE: <CBPeripheral: 0x2805a6a40, identifier = DEA8742F-E5B1-3879-EBB6-A07A7C92B485, name = PREOVA P1-C, mtu = 23, state = disconnected> can only accept commands while in the connected state
[FBP-iOS] handleMethodCall: startScan
[FBP-iOS] handleMethodCall: getConnectedSystemDevices
[FBP-iOS] handleMethodCall: getConnectedSystemDevices
flutter: Error reading RSSI: FlutterBluePlusException: readRssi: (code: 1) Timed out after 15s
[FBP-iOS] handleMethodCall: getConnectedSystemDevices
[FBP-iOS] handleMethodCall: stopScan
```

tsunhousam91, 2023-09-15:
Same Here.
In my case, I turned off, and turn on the adapter, and I want to reconnect the device and discover service.
but when I the app show the state is connected, and then, I try to discover service.  it will show this error:

API MISUSE: <CBPeripheral: 0x282a3e970, identifier = 4FF6376F-061B-F1E3-F59A-A271DF4063B0, name = DU_CT01, mtu = 23, state = disconnected> can only accept commands while in the connected state

chipweinberger, 2023-09-15:
this seems like a real bug in FBP. i'll investigate.

chipweinberger, 2023-09-15:
the changes made for hot restarting broke this.

you can look at the releases. somewhere in 1.14.xxx

chipweinberger, 2023-09-15:
i think this is the fix. use this code

```
- (void)disconnectAllDevices
{
    NSLog(@"[FBP-iOS] disconnectAllDevices");

    // request disconnections
    for (NSString *key in self.connectedPeripherals)
    {
        CBPeripheral *peripheral = [self.connectedPeripherals objectForKey:key];
        NSLog(@"[FBP-iOS] calling disconnect: %@", key);
        [self.centralManager cancelPeripheralConnection:peripheral];
    }

    // note: we do *not* clear self.knownPeripherals
    // Otherwise the peripheral would not have any strong references 
    // and would be garbage collected, making the didDisconnectPeripheral
    // callback not called

    [self.connectedPeripherals removeAllObjects];
    [self.servicesThatNeedDiscovered removeAllObjects];
    [self.characteristicsThatNeedDiscovered removeAllObjects];
    [self.didWriteWithoutResponse removeAllObjects];
    [self.peripheralMtu removeAllObjects];
}
```

chipweinberger, 2023-09-17:
fixed in 1.15.6

closed 2023-09-17 by chipweinberger