chipweinberger/flutter_blue_plus #565: [Bug]: Repeatedly receiving messages

xmw-simon, 2023-09-14:
### FlutterBluePlus Version

1.5.2

### Flutter Version

3.10.0

### What OS?

Android, iOS

### OS Version

all

### What happened?

when  I setNotifyValue(true), But this Bluetooth is abnormally disconnected，When reconnecting, you will receive duplicate messages， 
c.value.listen((value) {
            var str = utf8.decode(value);
            _dealWithData(str);
          }, cancelOnError: true);
so   I want to remove all listen for current ble,
what do i gotta be，
thanks 

### Logs

```shell
---->msg
---->msg
---->msg
```

chipweinberger, 2023-09-16:
seems like the same issue: https://github.com/boskokg/flutter_blue_plus/issues/559

logs are needed.

chipweinberger, 2023-09-16:
you need to stop listening to the stream (i.e. call `cancel`) when you disconnect

```
  final subscription = stream.listen((data) {
    print('Data: $data');
  });

  await subscription.cancel();
```

closed 2023-09-16 by chipweinberger

xmw-simon, 2023-09-18:
ok, let me try it, thanks