chipweinberger/flutter_blue_plus #567: [Bug]: PlatformException writeCharacteristic service not found on Android 12

lm699, 2023-09-15:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.13.2

### What OS?

Android, iOS

### OS Version

Android 12, Ios 16

### What happened?

When I try to write on device using c.write, I have exception appears. In IOS works fine.

cleanUUID function just 
`return uuid.toLowerCase().trim().replaceAll('\n', '');`

`Future<void> writeDataToDevice(
  BluetoothDevice device, 
  String serviceUUID, 
  String characteristicUUID, 
  List<int> dataToWrite
) async {
  if(connectState){
    List<BluetoothService> services = await device.discoverServices();
    for (var service in services) {
        String serviceUUIDFromDevice = cleanUUID(service.uuid.toString());
        String serviceUUIDClean = cleanUUID(serviceUUID);
        print('----- (fn Discover Services) Service UUID: ${service.uuid.toString()}');
        if (serviceUUIDFromDevice == serviceUUIDClean) {
              var characteristics = service.characteristics;
              for (var c in characteristics) {
                  print('----- Characteristic UUID: ${c.uuid.toString()}');
                  Debug = "Characteristic UUID: ${c.uuid.toString()}\n"+Debug;
                  if (c.uuid.toString() == characteristicUUID) {
                      await c.write(dataToWrite, withoutResponse: true);
                      print(c.uuid.toString());
                      print('------------------- Data written to device');
                      Debug = "{$dataToWrite} written to device\n"+Debug;
                      break;
                  }else{
                    print('------------------- Step 2 failed characteristic doesnt match');
                    Debug = "Step 2 failed characteristic doesnt match\n"+Debug;
                  }
              }
              break;
          } else {
              print('------------------- Step 1 failed services doesnt match');
              Debug = "Step 1 failed services doesnt match\n"+Debug;
          }
      }
  }
}`

The connection works fine, on 2 devices (android & iOS). The exception says : 
`PlatformException (PlatformException(writeCharacteristic, service not found '6400001-b5a3-f393-e0a9-e5024dccae', null, null))`

The function was called on ElevatedButton.

Thank for u help

### Logs

```shell
[... 99 lines cut ...]
I/flutter ( 7128): ------------------- Cleaning UUID...
I/flutter ( 7128): ----- (fn Discover Services) Service UUID: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
I/flutter ( 7128): ----- Characteristic UUID: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
I/flutter ( 7128): ------------------- Step 2 failed characteristic doesnt match
I/flutter ( 7128): ----- Characteristic UUID: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
D/[FBP-Android]( 7128): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt( 7128): discoverServices() - device: E2:ED:C2:76:3D:68
I/flutter ( 7128): 6e400002-b5a3-f393-e0a9-e50e24dcca9e
I/flutter ( 7128): ------------------- Data written to device
D/[FBP-Android]( 7128): [FBP-Android] onMethodCall: writeCharacteristic
D/BluetoothGatt( 7128): onSearchComplete() = Device=E2:ED:C2:76:3D:68 Status=0
D/[FBP-Android]( 7128): [FBP-Android] onServicesDiscovered: count: 4 status: 0
D/[FBP-Android]( 7128): [FBP-Android] onCharacteristicChanged: uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
D/BluetoothGatt( 7128): onPhyUpdate() - status=0 address=E2:ED:C2:76:3D:68 txPhy=2 rxPhy=2
D/BluetoothGatt( 7128): onPhyUpdate() - status=0 address=E2:ED:C2:76:3D:68 txPhy=2 rxPhy=1
D/BluetoothGatt( 7128): onPhyUpdate() - status=0 address=E2:ED:C2:76:3D:68 txPhy=2 rxPhy=2
D/BluetoothGatt( 7128): onPhyUpdate() - status=0 address=E2:ED:C2:76:3D:68 txPhy=2 rxPhy=1
D/BluetoothGatt( 7128): onPhyUpdate() - status=0 address=E2:ED:C2:76:3D:68 txPhy=2 rxPhy=2
Reloaded 2 of 802 libraries in 382ms.
E/BLASTBufferQueue( 7128): BLASTBufferItemConsumer::onDisconnect()
E/BLASTBufferQueue( 7128): BLASTBufferItemConsumer::onDisconnect()
Lost connection to device.

Exited.
```

chipweinberger, 2023-09-16:
your logs don't show that error message.

full logs needed

lm699, 2023-09-16:
Ok, Please see attached. This is a full logs of Visual Studio, where can I have full logs ?
For information , I tried with "WithoutResponse: true", same error.
[image]

chipweinberger, 2023-09-16:
does the example app work with that service?

lm699, 2023-09-16:
Yes, with the example code, after getting service, I can write : 

`D/[FBP-Android](12784): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt(12784): discoverServices() - device: E2:ED:C2:76:3D:68
D/BluetoothGatt(12784): onSearchComplete() = Device=E2:ED:C2:76:3D:68 Status=0
D/[FBP-Android](12784): [FBP-Android] onServicesDiscovered: count: 4 status: 0
D/[FBP-Android](12784): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](12784): [FBP-Android] onCharacteristicWrite: uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e status: 0
D/[FBP-Android](12784): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](12784): [FBP-Android] onCharacteristicWrite: uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e status: 0
D/[FBP-Android](12784): [FBP-Android] onMethodCall: writeCharacteristic
D/[FBP-Android](12784): [FBP-Android] onCharacteristicWrite: uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e status: 0
Application finished.`

Juste one parameter changed from my code : 
await c.write(_getRandomBytes(), withoutResponse: **c.properties.writeWithoutResponse**); | _true | false_

lm699, 2023-09-16:
I add await before my function, now it's works.

closed 2023-09-16 by lm699