chipweinberger/flutter_blue_plus #569: [Question]: Different subscription response from iOS & android.

DylanJCain, 2023-09-16:
### FlutterBluePlus Version

1.15.5

### Flutter Version

3.10.4

### What OS?

Android

### OS Version

Android 13

### Ask your question

So when I subscribe to a characteristic and write to it.

on iOS I get my expected response `{"query":"all","mac":"E0:5A:1B:50:C4:A2","host":"RAA-50C4A2","version":"5.3X","state":0,"result":"OK","ip":"0.0.0.0","rssi":0,"ble":9389010,"millis":9389010}`

on Andoird the response I get is: `{"query":"all","mac"}`

I just can't figure out why and wasn't sure where else to ask, I know it's a super specific question and only related to my use-case, I was just hoping for some suggestions.

### Logs

```shell
N/A - I'm not receiving any error as such, just an unexpected response.
```

chipweinberger, 2023-09-16:
seems like a bug in your code somewhere

DylanJCain, 2023-09-16:
@chipweinberger I thought so too, so I literally just cloned the example repo, and the only change I made was to send that query.

`final queryData = json.encode({"query": "all"});
    final queryBytes = utf8.encode(queryData);`

` onWritePressed: () async {
                      try {
                        await c.write(queryBytes, withoutResponse: c.properties.writeWithoutResponse);
                        final snackBar = snackBarGood("Write: Success");
                        snackBarKeyC.currentState?.removeCurrentSnackBar();
                        snackBarKeyC.currentState?.showSnackBar(snackBar);
                        if (c.properties.read) {
                          await c.read();
                        }
                      } catch (e) {
                        final snackBar = snackBarFail(prettyException("Write Error:", e));
                        snackBarKeyC.currentState?.removeCurrentSnackBar();
                        snackBarKeyC.currentState?.showSnackBar(snackBar);
                      }
                    },`

chipweinberger, 2023-09-16:
you probably need to request a higher mtu on android 

see readme

closed 2023-09-16 by chipweinberger