chipweinberger/flutter_blue_plus #570: [Bug]: FlutterBluePlusException: setNotifyValue: (code: 6) Timed out after 15s on IOS

hyeonjeong-roh, 2023-09-17:
### FlutterBluePlus Version

1.14.11

### Flutter Version

3.10.6

### What OS?

iOS

### OS Version

IOS 16.6

### What happened?

My FlutterBluePlus version isn't the latest, but I couldn't find any changes related to my issue in the changelog. I'm attempting to connect a pair of devices to my application, with each device having a unique ID. However, a timeout error occurs at this point:
```dart
 BmDiscoverServicesResult response = await futureResponse.fbpTimeout(timeout, "discoverServices");
``` 

I've structured my code to connect each device separately. 
Sometimes, discoverServices works well on both devices, but at other times, it functions correctly with only one device, while the other one encounters a timeout exception. 
Even when I increase the timeout to 100 seconds or more, the timeout exception still occurs.

I'm attempting to connect to each device using this approach:

```dart
  await FullScreenLoadingIndicator.show(
              newDevice.connectAndInitialize(),
            );       
``` 
```dart
 Future<bool> connectAndInitialize() async {
    if (state == BluetoothConnectionState.connected) {
      return true;
    }

    if (!(state == null || state == BluetoothConnectionState.disconnected)) {
      return false;
    }

    try {
      await device.connect(
        timeout: _connectionTimeout,
      );

      await initialize();

      return true;
    } catch (e, s) {
      Logger.error(
        'Error occurred while connecting to device',
        title: 'device bluetooth connection error',
        exception: e,
        stackTrace: s,
      );
      return false;
    }
  }
``` 
Inside the initialize() method is where I request discoverServices:
```dart
 Future<void> initialize() async {
    const customServiceUuid = '9A9E';
    try {
      final services = await device.discoverServices(); 
    
.....
``` 
I hope I didn't provide too much detail.

### Logs

```shell
Exception: FlutterBluePlusException
      
      ----- Start of Exceptioin Message -----
      Exception Message: FlutterBluePlusException: setNotifyValue: (code: 6) Timed out after 15s
      ----- End of Exception Message -----
      
      ----- Start of Stack Trace -----
#0      FutureTimeout.fbpTimeout.<anonymous closure>
utils.dart:59
#1      _rootRun (dart:async/zone.dart:1391:47)
#2      _CustomZone.run (dart:async/zone.dart:1301:19)
#3      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:882:34)
#4      _rootRun (dart:async/zone.dart:1391:47)
#5      _CustomZone.run (dart:async/zone.dart:1301:19)
#6      _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
#7      _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
#8      _rootRun (dart:async/zone.dart:1399:13)
#9      _CustomZone.run (dart:async/zone.dart:1301:19)
#10     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:1233:23)
#11     Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
#12     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
#13     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
#14     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:189:12)
      
      ----- End of Stack Trace -----
```

chipweinberger, 2023-09-17:
1. I would try updating
2. is`OnDiscoverServicesResponse` ever seen in the logs? I assume not

It could be a problem with your device

It could be a bad connection.

try the example app!

hyeonjeong-roh, 2023-09-18:
Thank you for your quick response! 

1. I'll try updating
2. My apologies for the earlier mistake. This is the log I intended to upload
```dart
 ----- Start of Exceptioin Message -----
      Exception Message: FlutterBluePlusException: discoverServices: (code: 6) Timed out after 15s
      ----- End of Exception Message -----
      
      ----- Start of Stack Trace -----
#0      FutureTimeout.fbpTimeout.<anonymous closure>
utils.dart:59
#1      _rootRun (dart:async/zone.dart:1391:47)
#2      _CustomZone.run (dart:async/zone.dart:1301:19)
#3      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:882:34)
#4      _rootRun (dart:async/zone.dart:1391:47)
#5      _CustomZone.run (dart:async/zone.dart:1301:19)
#6      _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
#7      _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
#8      _rootRun (dart:async/zone.dart:1399:13)
#9      _CustomZone.run (dart:async/zone.dart:1301:19)
#10     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:1233:23)
#11     Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
#12     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
#13     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
#14     _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:189:12)
      
``` 
3~5. I'll try the example app! 

I have a question. Considering that the device is currently connected, could a failed discoverServices request be due to a connection issue?

hyeonjeong-roh, 2023-09-22:
Thanks for keeping checking back!

I tried updating version to 1.15.7but it didn’t works for my case
In my situation, The discoverServices timeout error only occurs when attempting to pair two devices.
Therefore, I temporarily modified my code to connect the devices one by one. It works fine now!

chipweinberger, 2023-09-30:
@hyeonjeong-roh , can you share the code change you made?

chipweinberger, 2023-11-03:
@hyeonjeong-roh bump

chipweinberger, 2023-11-03:
fixed in 1.21.0

closed 2023-11-03 by chipweinberger