chipweinberger/flutter_blue_plus #573: [Bug]: Readable/writable characteristic is not readable/writable in Android

qawitherev, 2023-09-19:
### FlutterBluePlus Version

1.14.12

### Flutter Version

3.13.2

### What OS?

All

### OS Version

Android 12, iOS 16.4

### Bluetooth Module

Esp32

### What happened?

Readable/writable characteristic is not readable/writable in Android. Sample code: 

`void provisionTest() async {`
`await getServiceList();`
`getCharacteristics();`
`final ourCharacteristics = characteristicsList[0];`
`final canRead = ourCharacteristics.properties.read;`
`print("QAWI3 - can read: $canRead");`
`}`

On Android 12: 
[image]

On iOS 16.4, behaves like what expected: 
[image]

### Logs

```shell
Above picture
```

chipweinberger, 2023-09-19:
thanks for filing.

Can you add some logs to the android java code? to see if this issue is in android OS or fbp code

qawitherev, 2023-09-19:
Init BLE
[image]

Scanning
[image]

Connecting
[image]

Read Attempt
[image]

chipweinberger, 2023-09-19:
This is the java code

```
    HashMap<String, Object> bmCharacteristicProperties(int properties) {
        HashMap<String, Object> props = new HashMap<>();
        props.put("broadcast",                      (properties & 1)   != 0 ? 1 : 0);
        props.put("read",                           (properties & 2)   != 0 ? 1 : 0);
        props.put("write_without_response",         (properties & 4)   != 0 ? 1 : 0);
        props.put("write",                          (properties & 8)   != 0 ? 1 : 0);
        props.put("notify",                         (properties & 16)  != 0 ? 1 : 0);
        props.put("indicate",                       (properties & 32)  != 0 ? 1 : 0);
        props.put("authenticated_signed_writes",    (properties & 64)  != 0 ? 1 : 0);
        props.put("extended_properties",            (properties & 128) != 0 ? 1 : 0);
        props.put("notify_encryption_required",     (properties & 256) != 0 ? 1 : 0);
        props.put("indicate_encryption_required",   (properties & 512) != 0 ? 1 : 0);
        return props;
    }
```

chipweinberger, 2023-09-19:
You should also print the uuid to make sure you are comparing the same thing

print("QAWI3 - can read: $canRead uuid ${chr.uuid}");

qawitherev, 2023-09-19:
ok, found another weird thing, idk if this supposed to be happening
[image]

on ios, theres only 2 characteristics (as it should), but on android there's 3 of them with different uuid as its ios counterpart

chipweinberger, 2023-09-19:
ya that's fine. iOS handles some ble characteristics internally, whereas android exposes them.

closed 2023-09-19 by chipweinberger

chipweinberger, 2023-09-19:
ignore the characteristics you don't care about

qawitherev, 2023-09-19:
The characteristics that I want to interact with are ff01 and ff02, but in android, the service doesn't have these characteristics.