chipweinberger/flutter_blue_plus #574: [Bug]: Mutex Lock Does Not Release When Exception Occurs in

wookjaejoe, 2023-09-19:
### FlutterBluePlus Version

1.15.7

### Flutter Version

3.7.1

### What OS?

All

### OS Version

Android 14

### Bluetooth Module

ETC

### What happened?

For example, when a Bluetooth device is in a connected state and the discoverServices operation is in progress, if the connection is suddenly lost, an exception will occur at line 325 of the FlutterBluePlus._invokeMethod function, and mtx.give() will not be called.

After this case occurs, whenever FlutterBluePlus._invokeMethod is called, it always gets stuck at line 303. I believe that a modification is needed to ensure that mtx.give() is called correctly even when an exception occurs, using a try-finally statement.

### Logs

```shell
No log required.
```

chipweinberger, 2023-09-19:
you are right. thanks.

chipweinberger, 2023-09-19:
fixed 1.15.8

closed 2023-09-19 by chipweinberger