chipweinberger/flutter_blue_plus #576: [Bug]:

HuangWenJunGT, 2023-09-19:
### FlutterBluePlus Version

1.15.8

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

13

### Bluetooth Module

Nordic 52840

### What happened?

Calling the startScan method on an Android 13 phone will result in the following error. Unable to scan surrounding Bluetooth devices properly. But on phones under Android 13, they can be used normally。

### Logs

```shell
W/BluetoothAdapter(18400): getBluetoothLeScanner() ble not available
E/flutter (18400): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(startScan, getBluetoothLeScanner() is null. Is the Adapter on?, null, null)
E/flutter (18400): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (18400): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (18400): <asynchronous suspension>
E/flutter (18400): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:325:13)
E/flutter (18400): <asynchronous suspension>
E/flutter (18400): #3      FlutterBluePlus.startScan (package:flutter_blue_plus/src/flutter_blue_plus.dart:159:5)
E/flutter (18400): <asynchronous suspension>
```

chipweinberger, 2023-09-19:
permissions issue?

is bluetooth on?

ogak-github, 2023-09-20:
In my case, FBP stopped at getAdapterState. The value is null.
[image]
[image]

Tested on Android OS : 6.0.1

HuangWenJunGT, 2023-09-20:
I confirm that Bluetooth is enabled and declare the following permissions in AndroidManifest.xml:
uses permission android: name="android. permission. BLUETOOTH"
uses permission android: name="android. permission. BLUETOOTH_ADMIN"
uses permission android: name="android. permission. BLUETOOTH_SCAN"
uses permission android: name="android. permission. BLUETOOTH_CONNECT"
uses permission android: name="android. permission. BLUETOOTH_ADVERTISE"
uses permission android: name="android. permission. ACCESS_COARSE_LOCATION"
uses permission android: name="android. permission. ACCESS-FINE_LOCATION"

In addition, during runtime, the following permissions were dynamically applied for：
Permission.bluetoothScan,
Permission.bluetoothConnect,
Permission.bluetoothAdvertise,
Permission.locationWhenInUse,

The testing phones I use are Xiaomi 13 and Samsung S22

chipweinberger, 2023-09-20:
not sure. do other scanner apps work?

HuangWenJunGT, 2023-09-20:
I tried using several other Bluetooth libraries and reported error messages similar to the one I provided above.
I carefully reviewed the changes in Android 13 compared to 12, but did not find any significant changes in the Bluetooth section.
Can you try using an Android 13 phone to see if it can solve this problem

chipweinberger, 2023-09-25:
Did you figure it out?

HuangWenJunGT, 2023-09-26:
I'm very sorry, it's my problem. After multiple tests, it runs normally on Android 13.

chipweinberger, 2023-09-26:
what was the issue?

chipweinberger, 2023-09-30:
bump, what was the issue?

closed 2023-09-30 by chipweinberger