chipweinberger/flutter_blue_plus #578: [Question]:

Sumitdhiman-HHC, 2023-09-19:
### FlutterBluePlus Version

1.15.2

### Flutter Version

3.13.3

### What OS?

Android, iOS

### OS Version

iOS: 16.5.1

### Bluetooth Module

BLE device

### Ask your question

Android:
On android device we are able to find the localName from device json(object) and can go ahead according to our coding level condition.

iOS:
On iOS side localName is not coming like Android, it is showing different in device json (object) however it is showing in AdvertisementData(object) same as Android but after some time, not instantly. So we are not able to find the our BLE device.

On android side it is working perfectly.
What can be the issue here?
We followed same steps as mentioned in the official document.

### Logs

```shell
***** RESULT WHEN IT STARTS SCANNING THE DEVICES *****

ScanResult{device: BluetoothDevice{remoteId: FC196CF0-3840-A282-2E79-8DDCC915E82D, localName: ts7680-526571, type: BluetoothDeviceType.le, isDiscoveringServices: false, services: null}, advertisementData: AdvertisementData{localName: , txPowerLevel: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [6E400001-B5A3-F393-E0A9-E50E24DCCA9E]}, rssi: -77, timeStamp: 2023-09-19 20:35:48.014587}

***** RESULT AFTER SOME TIME WITH ADVERTISEMENTDATA LOCALNAME ******

ScanResult{device: BluetoothDevice{remoteId: 94366FFB-FE18-33D4-5079-8FB934FB3773, localName: ts7680-4ceff4, type: BluetoothDeviceType.le, isDiscoveringServices: false, services: null}, advertisementData: AdvertisementData{localName: UVDI_59997, txPowerLevel: null, connectable: true, manufacturerData: {}, serviceData: {}, serviceUuids: [6E400001-B5A3-F393-E0A9-E50E24DCCA9E]}, rssi: -65, timeStamp: 2023-09-19 20:35:48.230487}
```

chipweinberger, 2023-09-19:
after connection, iOS uses the GAP name.

this is an annoying issue and FBP should fix it imo.

Sumitdhiman-HHC, 2023-09-21:
Yes, I mean to say which name I'm getting from device.localName while scanning on Android Device then it is giving the correct name in variable device.localName but for iOS it is giving the same name in advertisementData.localName instead of device.localName

So the question is: We need to use device.localName for Android and advertisementData.localName for iOS, right ?

chipweinberger, 2023-09-28:
> So the question is: We need to use device.localName for Android and advertisementData.localName for iOS, right ?

correct. this is a known issue. on iOS `device.localName` is the GAP name. this should be fixed.

chipweinberger, 2023-09-28:
fixed da0ed962c20a731cf057f287f93815e536e48103 

    [Fix] localName: iOS should use advertisement name to match Android

chipweinberger, 2023-09-28:
fixed 1.15.10

closed 2023-09-28 by chipweinberger

Sumitdhiman-HHC, 2023-09-28:
Thankyou

chipweinberger, 2023-11-04:
in 1.25.1 we now expose all the various name