chipweinberger/flutter_blue_plus #579: [Bug]: Unhandled Exception: FlutterBluePlusException: discoverServices: (code: 1) Timed out after 30s

amirsuwal01, 2023-09-20:
### FlutterBluePlus Version

^1.15.1

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

10

### Bluetooth Module

Nordic

### What happened?

I tried with a time of 15 and 30 but was not able to discover the services...Sometimes it does discover services in a defined time. I connect the BLE directly via the Device address.

### Logs

```shell
E/flutter (10797): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException: discoverServices: (code: 1) Timed out after 30s
E/flutter (10797): #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:57:7)
E/flutter (10797): #1      _RootZone.run (dart:async/zone.dart:1655:54)
E/flutter (10797): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:882:34)
E/flutter (10797): #3      Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
E/flutter (10797): #4      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
E/flutter (10797): #5      _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
E/flutter (10797): #6      _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:189:12)
```

chipweinberger, 2023-09-20:
update to 1.15.8

Palaceapp, 2023-09-22:
Thanks for reply.. The issue was because of not disconneting the device properly..it discovered services as the bluetooth is disconnected properly

amirsuwal01, 2023-09-22:
The issue is resolved once the ble is disconnected properly.

closed 2023-09-22 by amirsuwal01

cl0v, 2023-11-09:
@amirsuwal01 What do you mean by disconnect properly?

chipweinberger, 2023-11-09:
he is referring to a bug in FBP that has been fixed

FBP was not updating its internal state, so it thought it was still connected but it wasn't

ProductOfAmerica, 2024-02-06:
@chipweinberger Possible this is still happening? I was running perfectly fine with my BLE arduino code, then out of nowhere started getting these errors:

```
E/flutter ( 3650): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException | writeCharacteristic | fbp-code: 1 | Timed out after 15s
E/flutter ( 3650): #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:71:7)
E/flutter ( 3650): #1      _RootZone.run (dart:async/zone.dart:1655:54)
E/flutter ( 3650): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:942:34)
E/flutter ( 3650): #3      Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18:15)
E/flutter ( 3650): #4      _Timer._runTimers (dart:isolate-patch/timer_impl.dart:398:19)
E/flutter ( 3650): #5      _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:429:5)
E/flutter ( 3650): #6      _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:184:12)
```

c0c4i, 2024-11-07:
@chipweinberger any news on that? @ProductOfAmerica did you find a solution?

ProductOfAmerica, 2024-11-07:
> @chipweinberger any news on that? @ProductOfAmerica did you find a solution?

Nope :(

chipweinberger, 2024-11-07:
it's most likely a problem with your peripheral, not FBP. I would start there. look at it's logs.