chipweinberger/flutter_blue_plus #581: [Bug]: dyld[2179]: Symbol not found: _OBJC_CLASS_$_FlutterError

kbessemer, 2023-09-20:
### FlutterBluePlus Version

1.4.0

### Flutter Version

3.13.4

### What OS?

iOS

### OS Version

iOS 15.7.9 & iOS 17

### Bluetooth Module

STM32

### What happened?

The flutter app will not start on iOS, it crashes immediately.

### Logs

```shell
dyld[2179]: Symbol not found: _OBJC_CLASS_$_FlutterError
  Referenced from: <D26B8D55-5052-3B29-A2D1-64F3A74A6628> /private/var/containers/Bundle/Application/54180268-0E54-41B4-A5ED-E03F662FAE86/Runner.app/Runner
  Expected in:     <6DD9F429-0BD2-3581-B55F-23B9A86E1CC2> /private/var/containers/Bundle/Application/54180268-0E54-41B4-A5ED-E03F662FAE86/Runner.app/Frameworks/flutter_blue_plus.framework/flutter_blue_plus
```

kbessemer, 2023-09-20:
I resolved this bug by upgrading my Flutter Blue Plus plugin to 1.15.8

closed 2023-09-21 by chipweinberger

Yongle-Fu, 2023-09-21:
https://stackoverflow.com/questions/77144171/dyld2179-symbol-not-found-objc-class-fluttererror/77149456#77149456

Yongle-Fu, 2023-09-21:
same issue, anything reason ?

kbessemer, 2023-09-21:
> same issue, anything reason ?

upgrade your plugin to 1.15.8

Yongle-Fu, 2023-09-22:
Thanks, Could you please tell me the specific reason?

Yongle-Fu, 2023-09-27:
found reason, then will be worked on Xcode15
1. [Protobuf] remove protobuf from iOS and MacOS
2. remove permission_handler dependency

vinicz, 2023-10-03:
> found reason, this commit fixed the issue: [[Protobuf] remove protobuf from iOS and MacOS](https://github.com/boskokg/flutter_blue_plus/commit/1848617e6ce25385c2a7b4129aa199801bcfee71)
> 
> I use flutter_blue in my code (not plus), so these are my options to fix the error:
> 
> 1. migrate the flutter_blue_plus
> 2. refactor to swift protobuf also worked.
> 3. remove permission_handler dependency

@Yongle-Fu the issue is not that bad, found a hint here https://developer.apple.com/forums/thread/735610, the main cause is that the old flutter_blue code still catches and throws FlutterErrors, this was changed here as well. After we got rid of that in our fork everything worked again. (But would still recommend migrating to this plugin at some point)

Yongle-Fu, 2023-10-07:
Thank you for your help, i had update the objective c code to swift code and remove the try catch block indeed.

chipweinberger, 2023-11-27:
so to summarize, a simple fix is to fork the old FBP and remove any catching or throwing of FlutterError. this was invalid code. It should be NSException. see master branch for details. i.e.

```
    @catch (NSException *e)
    {
        NSString *stackTrace = [[e callStackSymbols] componentsJoinedByString:@"\n"];
        NSDictionary *details = @{@"stackTrace": stackTrace};
        result([FlutterError errorWithCode:@"iosException" message:[e reason] details:details]);
    }
```