chipweinberger/flutter_blue_plus #582: [Bug]: "FlutterBluePlus.turnOn()" will crash when Bluetooth is not authorized on Android12.

shuizhuyu1015, 2023-09-21:
### FlutterBluePlus Version

1.15.8

### Flutter Version

3.3.2

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

Nordic nRF5

### What happened?

First of all, my phone is **Samsung Android12**, and the Bluetooth adapter is **not turned on**.
After launching the app, calling "**FlutterBluePlus.turnOn()**" will first bring up whether to allow connection to nearby devices authorization dialog, **select disable**, and then the app **crashes**!

### Logs

```shell
[FBP-Android] onMethodCall: flutterHotRestart
[FBP-Android] initializing BluetoothAdapter
[FBP-Android] disconnectAllDevices
[FBP-Android] closeAllDevices
[FBP-Android] connectedPeripherals: 0
[FBP-Android] HotRestart: complete
[FBP-Android] onMethodCall: getAdapterState
[FBP-Android] onMethodCall: setLogLevel
[FBP-Android] onMethodCall: turnOn

D/AndroidRuntime(21085): Shutting down VM
E/AndroidRuntime(21085): FATAL EXCEPTION: main
E/AndroidRuntime(21085): java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1452, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord (pid=21085, uid=10767) requires android.permission.BLUETOOTH_CONNECT
E/AndroidRuntime(21085): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:6552)
E/AndroidRuntime(21085): 	at android.app.ActivityThread.handleSendResult(ActivityThread.java:6591)
E/AndroidRuntime(21085): 	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
E/AndroidRuntime(21085): 	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:48)
E/AndroidRuntime(21085): 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:149)
E/AndroidRuntime(21085): 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:103)
E/AndroidRuntime(21085): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2931)
E/AndroidRuntime(21085): 	at android.os.Handler.dispatchMessage(Handler.java:117)
E/AndroidRuntime(21085): 	at android.os.Looper.loopOnce(Looper.java:205)
E/AndroidRuntime(21085): 	at android.os.Looper.loop(Looper.java:293)
E/AndroidRuntime(21085): 	at android.app.ActivityThread.main(ActivityThread.java:9685)
E/AndroidRuntime(21085): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(21085): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:586)
E/AndroidRuntime(21085): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1211)
E/AndroidRuntime(21085): Caused by: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord (pid=21085, uid=10767) requires android.permission.BLUETOOTH_CONNECT
E/AndroidRuntime(21085): 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2443)
E/AndroidRuntime(21085): 	at android.os.Parcel.createException(Parcel.java:2427)
E/AndroidRuntime(21085): 	at android.os.Parcel.readException(Parcel.java:2410)
E/AndroidRuntime(21085): 	at android.os.Parcel.readException(Parcel.java:2352)
E/AndroidRuntime(21085): 	at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2384)
E/AndroidRuntime(21085): 	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1769)
E/AndroidRuntime(21085): 	at android.app.Activity.startActivityForResult(Activity.java:5630)
E/AndroidRuntime(21085): 	at android.app.Activity.startActivityForResult(Activity.java:5554)
E/AndroidRuntime(21085): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$0$com-boskokg-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:322)
E/AndroidRuntime(21085): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda1.op(Unknown Source:4)
E/AndroidRuntime(21085): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.lambda$askPermission$5$com-boskokg-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:1219)
E/AndroidRuntime(21085): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda0.op(Unknown Source:6)
E/AndroidRuntime(21085): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onRequestPermissionsResult(FlutterBluePlusPlugin.java:1179)
E/AndroidRuntime(21085): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:777)
E/AndroidRuntime(21085): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:409)
E/AndroidRuntime(21085): 	at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:757)
E/AndroidRuntime(21085): 	at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:748)
E/AndroidRuntime(21085): 	at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:8988)
E/AndroidRuntime(21085): 	at android.app.Activity.dispatchActivityResult(Activity.java:8845)
E/AndroidRuntime(21085): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:6545)
E/AndroidRuntime(21085): 	... 13 more
E/AndroidRuntime(21085): Caused by: android.os.RemoteException: Remote stack trace:
E/AndroidRuntime(21085): 	at com.android.server.wm.ActivityTaskSupervisor.checkStartAnyActivityPermission(ActivityTaskSupervisor.java:1244)
E/AndroidRuntime(21085): 	at com.android.server.wm.ActivityStarter.executeRequest(ActivityStarter.java:1302)
E/AndroidRuntime(21085): 	at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:855)
E/AndroidRuntime(21085): 	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1383)
E/AndroidRuntime(21085): 	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1355)
E/AndroidRuntime(21085): callee: null 1733/2045
E/AndroidRuntime(21085):
```

timalenustmf, 2023-09-21:
Interesting. I am experiencing similar issues. 
Before we call turnOn() we check if all needed permissions are granted using the package permission_handler. I was under the impression that this exception was coming from this package. Hence why I opened https://github.com/Baseflow/flutter-permission-handler/issues/1153
Do you also use permission_handler @shuizhuyu1015 ? Or solely flutter_blue_plus?

Judging from our crash logs it does not seem to happen for all Android 12 devices, mainly Samsung.

shuizhuyu1015, 2023-09-21:
> 有趣的。我遇到类似的问题。 在调用turnOn()之前，我们检查是否使用permission_handler包授予了所有需要的权限。我的印象是这个异常来自这个包。这就是为什么我打开[Baseflow/flutter-permission-handler#1153](https://github.com/Baseflow/flutter-permission-handler/issues/1153) 你也使用permission_handler吗@shuizhuyu1015？或者只是 flutter_blue_plus？
> 
> 从我们的崩溃日志来看，似乎并不是所有 Android 12 设备（主要是三星）都会发生这种情况。

Thank you for your reply.

My app does access "permission_handler", but it doesn't call it when Bluetooth is started.

That is, when calling turnOn, flutter_blue_plus will first pop out the authorization dialog, if you select allow, it will display the open adapter dialog, and if you select disable, it will crash directly.

All three of my Android12 phones (Samsung/HUAWEI/OPPO) will crash.

timalenustmf, 2023-09-21:
@shuizhuyu1015 what happens if you first ensure you have the BLUETOOTH_CONNECT permission using the "permission_handler" package and then invoke the turnOn of flutter_blue_plus? As the permission should already be granted this turnOn should not dislplay the authorization dialog anymore. 

I'll try to test this case myself as well.

shuizhuyu1015, 2023-09-21:
> @shuizhuyu1015 what happens if you first ensure you have the BLUETOOTH_CONNECT permission using the "permission_handler" package and then invoke the turnOn of flutter_blue_plus? As the permission [...]

I tried to check the permissions with "permission_handler" before "turnOn", and then call "turnOn" after the permissions were allowed, which solved the crash problem.Thanks!

timalenustmf, 2023-09-21:
Looking into the android implementation for `turnOn` it seems the permission is requested "the old way". 
- https://stackoverflow.com/questions/62671106/onactivityresult-method-is-deprecated-what-is-the-alternative/63654043#63654043
- https://stackoverflow.com/questions/68095709/startactivityforresult-is-deprecated-and-unable-to-request-for-bluetooth-connec

It seems `startActivityForResult` is deprecated. 

I am by no means a native Android developer, so I don't feel comfortable at the moment to implement a fix based on the StackOverFlow posts linked above. But maybe someone more knowledgeable can based on the information in this comment.

chipweinberger, 2023-10-10:
I cannot reproduce this cash using the example app.
* Android 13
* Pixel 5a

timalenustmf, 2023-10-11:
I can reproduce this crash with the example app

- Android 12
- Fairphone 4

Steps to reproduce:
1. turn off bluetooth
2. run example app
3. press "turn on" button in the example app
4. example app crashes

<details><summary>Here are the logs from the debug console</summary>

```

flutter run
Launching lib/main.dart on FP4 in debug mode...
Running Gradle task 'assembleDebug'...                              4.5s
✓  Built build/app/outputs/flutter-apk/app-debug.apk.
Syncing files to device FP4...                                     130ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

An Observatory debugger and profiler on FP4 is available at: http://127.0.0.1:40685/J-FoHYnSb_E=/
D/[FBP-Android]( 5167): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android]( 5167): [FBP] initializing BluetoothAdapter
D/[FBP-Android]( 5167): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android]( 5167): [FBP] connectedPeripherals: 0
D/[FBP-Android]( 5167): [FBP] onMethodCall: getAdapterState
The Flutter DevTools debugger and profiler on FP4 is available at: http://127.0.0.1:9102?uri=http://127.0.0.1:40685/J-FoHYnSb_E=/
D/[FBP-Android]( 5167): [FBP] onMethodCall: turnOn
D/AndroidRuntime( 5167): Shutting down VM
E/AndroidRuntime( 5167): FATAL EXCEPTION: main
E/AndroidRuntime( 5167): Process: com.charlesweinberger.example, PID: 5167
E/AndroidRuntime( 5167): java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1452, result=0, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.charlesweinberger.example/com.charlesweinberger.example.MainActivity}: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{365791c 5167:com.charlesweinberger.example/u0a12} (pid=5167, uid=10012) requires android.permission.BLUETOOTH_CONNECT
E/AndroidRuntime( 5167):        at android.app.ActivityThread.deliverResults(ActivityThread.java:5302)
E/AndroidRuntime( 5167):        at android.app.ActivityThread.handleSendResult(ActivityThread.java:5341)
E/AndroidRuntime( 5167):        at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
E/AndroidRuntime( 5167):        at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
E/AndroidRuntime( 5167):        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
E/AndroidRuntime( 5167):        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
E/AndroidRuntime( 5167):        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2211)
E/AndroidRuntime( 5167):        at android.os.Handler.dispatchMessage(Handler.java:106)
E/AndroidRuntime( 5167):        at android.os.Looper.loopOnce(Looper.java:201)
E/AndroidRuntime( 5167):        at android.os.Looper.loop(Looper.java:288)
E/AndroidRuntime( 5167):        at android.app.ActivityThread.main(ActivityThread.java:7894)
E/AndroidRuntime( 5167):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 5167):        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
E/AndroidRuntime( 5167):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1009)
E/AndroidRuntime( 5167): Caused by: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{365791c 5167:com.charlesweinberger.example/u0a12} (pid=5167, uid=10012) requires android.permission.BLUETOOTH_CONNECT
E/AndroidRuntime( 5167):        at android.os.Parcel.createExceptionOrNull(Parcel.java:2426)
E/AndroidRuntime( 5167):        at android.os.Parcel.createException(Parcel.java:2410)
E/AndroidRuntime( 5167):        at android.os.Parcel.readException(Parcel.java:2393)
E/AndroidRuntime( 5167):        at android.os.Parcel.readException(Parcel.java:2335)
E/AndroidRuntime( 5167):        at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2284)
E/AndroidRuntime( 5167):        at android.app.Instrumentation.execStartActivity(Instrumentation.java:1745)
E/AndroidRuntime( 5167):        at android.app.Activity.startActivityForResult(Activity.java:5403)
E/AndroidRuntime( 5167):        at android.app.Activity.startActivityForResult(Activity.java:5361)
E/AndroidRuntime( 5167):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$onMethodCall$0$com-lib-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:338)
E/AndroidRuntime( 5167):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda1.op(Unknown Source:4)
E/AndroidRuntime( 5167):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.lambda$askPermission$5$com-lib-flutter_blue_plus-FlutterBluePlusPlugin(FlutterBluePlusPlugin.java:1245)
E/AndroidRuntime( 5167):        at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$$ExternalSyntheticLambda0.op(Unknown Source:6)
[... cut ...]
E/AndroidRuntime( 5167):        at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:669)
E/AndroidRuntime( 5167):        at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1201)
E/AndroidRuntime( 5167):        at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1173)
E/AndroidRuntime( 5167): 
I/Process ( 5167): Sending signal. PID: 5167 SIG: 9
Lost connection to device.
```

</details>

chipweinberger, 2023-10-11:
> requires android.permission.BLUETOOTH_CONNECT

In this java code I think we need to check 'granted' and return an error if false

`ensurePermissions(permissions, (granted, perm) -> {`

can you try it?

chipweinberger, 2023-10-11:
pushed a fix. try master branch.

timalenustmf, 2023-10-11:
I just tried it as well. I confirm, this fixes the crash.